raspberrypi/rpi-image-gen #7: Copying files from ${IGconf_ext_dir} into final rootfs user home directory during chroot

jpiwek, 2025-01-23:
Building the rootfs:
I created a customize-hooks entry in my meta/test.yaml file to copy some specific files from ${IGconf_ext_dir} into the final rootfs user home directory /home/$IGconf_first_user_name/ with correct owner settings.
I tried different approaches. Very often I encountered issues with permission.

My final approach which worked well, however I am afraid it is too complicated , looking forward to your comments if there is a simpler solution to that:

Example:

1. copying files of intersted into a tmp rootfs folder:
  ```
 install -m 755 -v -D -t $1/home/$IGconf_first_user_name/tmp/.ssh $IGconf_dir_ssh
authorized_keys
```
2.using chroot with userspec to copy files into home directory with right owner and mode
```
user=${IGconf_first_user_name}
install -m 755 -v -D -t $1/home/$user/tmp/.ssh $IGconf_dir_ssh/authorized_keys
    #install
    chroot --userspec=${user}:${user} $1 bash -- <<- EOCHROOT
    touch /home/$user/ipgw2.INSTALLED
    install -v -D -t /home/$user/.ssh /home/$user/tmp/.ssh/authorized_keys -m 600 
    
    EOCHROOT
```
ls -la results in:
`
-rw------- 1 100999 100999  185 Jan 23 16:53 authorized_keys
`

jpiwek, 2025-01-24:
Ok I do understand: copying files with correct ownership permissions I solved that way:

Install user files from build env into tmp folder in the target rootfs defined in $1:
`install -v -D -t $1/home/$user/tmp/.ssh $IGconf_dir_ssh/authorized_keys`

Starting a chroot session with --userspec parameter which includes target $user:
Touch then creates a file in home/$user directory with $user ownership
install copies authorized_keys with correct $user ownership and 600 permissions into $user home directory
   ```
 chroot --userspec=${user}:${user} $1 bash -- <<- EOCHROOT
    touch /home/$user/test.INSTALLED
    install -v -D -t /home/$user/.ssh /home/$user/tmp/.ssh/authorized_keys -m 600

    EOCHROOT 
```

hope that help

learmj, 2025-01-24:
Firstly, thank you for giving rpi-image-gen a try. Feedback like these early tickets is critically important and I welcome it. I'm determined to make the tool a solid alternative for developing and building products with and for it to have the flexibility and control that developers need to work efficiently. Please don't hold back - all feedback is welcome :)

Thanks for sharing your solution. I've tried to replicate what you're doing using ```examples/custom_layers/``` and I don't see the same problem. In YAML I can copy a file from ```IGconf_ext_nsdir``` into the chroot and chown it to the local user. After I build the image, write it to an sd card and boot it, the permissions are correct for the file that was chown'd:
```
acmedev@acme-devhost:~$ ls -l
total 4
-rw-r--r-- 1 root    root     0 Jan 24  2025 acme.INSTALLED
-rw-r--r-- 1 acmedev acmedev 77 Jan 24  2025 setup-functions
acmedev@acme-devhost:~$ 
```
Can you try the patch below and see if you observe the same?
```
--- a/examples/custom_layers/acme/meta/acme-sdk-v1.yaml
+++ b/examples/custom_layers/acme/meta/acme-sdk-v1.yaml
@@ -13,12 +13,13 @@ mmdebstrap:
   - cp $IGconf_ext_nsdir/setup-functions $1/home/$IGconf_first_user_name
   - |-
     chroot $1 bash -- <<- EOCHROOT
+    chown ${IGconf_first_user_name}:${IGconf_first_user_name} /home/${IGconf_first_user_name}/setup-functions
     source /home/$IGconf_first_user_name/setup-functions
     install_secrets $IGconf_first_user_name $IGconf_acme_dev_server
     touch /home/$IGconf_first_user_name/acme.INSTALLED
     EOCHROOT
-  cleanup-hooks:
-    - shred --verbose -u --zero $1/home/$IGconf_first_user_name/setup-functions
+    #  cleanup-hooks:
+    #- shred --verbose -u --zero $1/home/$IGconf_first_user_name/setup-functions
```

jpiwek, 2025-01-24:
It's been my pleasure. 
In my opinion rpi-image-gen is a great and neat tool which enables me to create fully customized, scaled and reproducable images with highest level of flexibilty. 
In the past I created lot of static scripts which required to be updated on new RPi OS release, which was very time consuming. You helped me a lot.
You can close this case as it gave me a good view on that.

I compared both example  which delivered similar results for me: 
with install I copy files to a tmp location in user rootfs home directory.
then I execute a chroot environment with --userspec flag which sets automatically the right ownership e.g. when creating files / directories.
install has some advantages over cp... as it creates automatically a directory, set ownership, rwx permission...
here is the full example which worked for me

```
customize-hooks:
    # Could perform ops here after the chroot is set up and all packages got installed, eg
  # - cp $IGconf_ext_nsdir/setup-functions $1/home/$IGconf_first_user_name
  - |-
    user=${IGconf_first_user_name}
    cp $IGconf_ext_nsdir/setup-functions $1/home/$IGconf_first_user_name
     #dir_ssh=${IGconf_ext_dir}/src/ssh -> defined in my.options file
    install -v -D -t $1/home/$user/tmp/.ssh $IGconf_dir_ssh/authorized_keys
    #install
    chroot --userspec=${user}:${user} $1 bash -- <<- EOCHROOT
    source /home/$IGconf_first_user_name/setup-functions
    install_secrets $IGconf_first_user_name $IGconf_ipgw2_dev_server
    touch /home/$user/ipgw2.INSTALLED
    install -v -D -t /home/$user/.ssh /home/$user/tmp/.ssh/authorized_keys -m 600 
    
    EOCHROOT
    
  cleanup-hooks:
    -  |- 
      rm -rf $1/home/$IGconf_first_user_name/tmp
      # shred --verbose -u --zero $1/home/$IGconf_first_user_name/setup-functions  
```

learmj, 2025-01-27:
Thank you @jpiwek Much appreciated. Good to hear the project is viewed as heading in the right direction. Closing.

closed 2025-01-27 by learmj