raspberrypi/rpi-image-gen #12: Slim example with util-linux

procount, 2025-02-05:
I'm having another play with the examples, particularly the `slim` example.
I notice that one apt package that is installed is `util-linux`. I assume this is the same package as used in buildroot with the same name. If so, it should install sfdisk and fdisk, but I can't find them in the resultant image. How do I get these tools installed in the image?

learmj, 2025-02-06:
Hi. Thanks for your report. This is a 'Debian thing'. There was obviously a decision made to remove fdisk from util-linux and put it in its own package for installation via apt. The ```fdisk``` package will give you what you need. ```util-linux``` doesn't provide it:
```
$ curl -sL -o- "http://deb.debian.org/debian/pool/main/u/util-linux/util-linux_2.38.1-5%2bdeb12u3_arm64.deb" | dpkg-deb -c /dev/stdin | grep fdisk
$
```
```
$ curl -sL -o- "http://deb.debian.org/debian/pool/main/u/util-linux/fdisk_2.38.1-5%2bdeb12u3_arm64.deb" | dpkg-deb -c /dev/stdin | grep fdisk
...
-rwxr-xr-x root/root    198848 2024-11-21 20:01 ./sbin/fdisk
-rwxr-xr-x root/root    198816 2024-11-21 20:01 ./sbin/sfdisk
...
```
rpi-image-gen is intended to be a tool that offers a high degree of customisation and control over the build. At the moment, there is no meta layer that installs fdisk and while this may seem odd, it's because I didn't want to start including programs in base layers and profiles etc that could be inherited/used by users when they don't need those programs.

I envisage rpi-image-gen being used by mainly developers who care enough about their system to want to create their own profile(s) and meta layers, and obviously are welcome to utilise the 'in-tree' ones, but their goal should only be to install what they need while being able to easily customise things they way they need to.

I expect a similar ticket regarding ```fsck``` which is not installed by default via any layer for the sole reason that it only makes sense if you need to check your fs, which may not apply in the case of a read-only filesystem deployment. Hope that helps to give some colour to the project.

And lastly, thank you for taking the time to give rpi-image-gen a try. I am determined to make it a solid 'go-to' tool for people to use, so please do let me know if you have any suggestions. All feedback is welcome. 

Cheers,
--  Matt

procount, 2025-02-06:
Thanks for the info, Matt.

I hadn't realised that Debian had split the fdisk utilities out of util-linux for apt packaging. The latest description output from `apt info` is not so explicit about what it includes and the source homepage for util-linux still includes fdisk/sfdisk et al, so I was confused.

I agree with your sentiments of keeping it lean and mean, which is why I was starting with the slim example. I have a particular use case in mind where I need a replacement base system of kernel + rootfs to host my application with the latest tools, and rapsios-lite is about 3-4 times too big.

I'll add the fdisk package and see where it takes me.

learmj, 2025-02-06:
I'm happy to help you with that so if you need anything w.r.t. rpi-image-gen that would make your task easier, drop me a line.

procount, 2025-02-07:
I now have the fdisk package added and sfdisk is now present.
Some sfdisk operations require root privileges, so I added the sudo package.
But how can I set up the pi user to be a member of the sudo group and/or add it to the sudoers file?
I guess I need a hook function somewhere?

(ASIDE: I'm mostly a procedural coder, so until I've got my head around the process flow of this tool, I shall struggle with it's declarative style of config files and their consequences. But that's just me. )

learmj, 2025-02-07:
Hi. Meta layer https://github.com/raspberrypi/rpi-image-gen/blob/master/meta/rpi/user-credentials.yaml does this for you.

procount, 2025-02-07:
Thanks, that fixed it, although I had to remove slim-customisations.yaml as well. Slowly getting there ;)

procount, 2025-02-07:
I know these examples are mostly Pi5 based, but what's the reason the Slim example won't run on a Pi4, even with arm_64bit=1 set in config.txt? It gets to the 4 raspberries and just sits there. Even the blinking cursor stops flashing after a while.
All the correct 64-bit DTBs seem to be present.
Adding a tty console allows it to show the login prompt but the keyboard is not functional. Is it missing some Pi4 USB drivers or something? Uart output seems normal with no errors reported until it stops after changing the serial port baud rate. No login prompt on there.

The good news is I can get my App to run on this image on a Pi5. It seems a bit erratic, but I still need to add some other packages and eventually convert the rootfs partition to a squashfs file to be stored on the BOOT partition.

learmj, 2025-02-07:
Not sure. I'll look into it. It could indeed be kernel boot args ```tty=``` related (just a guess).
I haven't tested images on anything other that pi5 and cm5 so far. I do intend to. FWIW I plan to add device dirs for cm5, pi4 and cm4 and any other device classes we intend to support with this. I'll need to have fleshed things out by then.
I'm pleased you've made some progress.

procount, 2025-02-07:
Eventually, I'll need an image that supports all RPi models, like Raspios-lite does. :wink:

jonnymacs, 2025-02-11:
@procount I've been working with this utility for about a month now and I've had success wrangling things to build images for the pi4.

The slim image will boot on the pi4 if you update /examples/slim/device/mypi5/device/rootfs-overlay/boot/firmware/cmdline.txt to match what's in the /templates/rpi/boot-firmware/cmdline.txt

I don't exactly know why, but here is the diff

`@@ -1 +1 @@
-console=ttyAMA10,115200 root=/dev/disk/by-label/ROOT rootfstype=ext4 fsck.repair=yes rootwait
+console=serial0,115200 console=tty1 root=ROOTDEV rootfstype=ext4 fsck.repair=yes rootwait`

cc: @learmj 

I have some more learnings as well not sure where to share, I'm using Docker to to build images in a debian:bookworm base container on a mac, then I use the rpi imager tool to load the generated img on a usb stick to test on the pi4. I upgraded a few weeks ago from an old intel to an m4 mac which required some slight adjustments, and a few updates to the master branch here I've had to keep up with, but overall I've got a pretty nice workflow at this point if others are interested.

jonnymacs, 2025-02-11:
@procount forgive me - i did not read your entire post - I see you tried this. It boots but yes, and sits there. My keyboard works though.

learmj, 2025-02-11:
Hi @jonnymacs Thank you for also giving it a try and I'm glad to hear it's working for you. I did email some known users a little while ago just to let them know about a backward incompatible change that I intended to make. Hope that didn't disrupt you too much.

Re: pi4, I do intend to get this in shape at some point soon. The console compatibility can probably be fixed simply by using a rootfs-overlay for the pi4 device class (eg ```pi4/device/rootfs-overlay/boot/firmware/cmdline.txt```).

jonnymacs, 2025-02-11:
@learmj all good on the change. `board -> device`. Makes sense. I always check the commit logs before a git pull as I know this is an active project 😄 

Is there a discord / slack / working group somewhere for this project?

procount, 2025-02-11:
@jonnymacs, - encouraged, I tried again. It turns out my USB SD card reader is not bootable on the Pi4. It booted fine in the SD card slot, and again with a different USB SD card reader.

learmj, 2025-02-12:
@jonnymacs Very sensible :) FYI I plan to make some tweaks to variable handling in the next day or so which will likely (small) cause issues. I'll make things clear in the commit msg.

> Is there a discord / slack / working group somewhere for this project?

No

closed 2025-02-26 by learmj