raspberrypi/rpi-image-gen #13: Feature request: btrfs

solskogen, 2025-02-07:
Add support for using btrfs instead of ext4.

learmj, 2025-02-07:
Hi. Thanks for your feedback (I'll post back on the forum post a little bit later). I am open to this but it's important to point out that the in-tree image layouts may not suit everyone and I did not want to deviate too far from what RPiOS provides today (certainly for the 'simple' layout). I wanted the tool to be easy to use in order that someone can make as little or as many of their own own customisations as they wished.

As an example, to create an identical 1 x vfat and 1 x btrfs layout it would be as simple as:

1. Use ```-D``` to specify a path to an external directory at build time
2. In that external directory, create your image dir (keeping the same ```mbr/simple_dual``` name would be fine and would mean no config file changes to the default).
3. Copy everything from the in-tree ```mbr/simple_dual``` to ```<EXT_DIR>/mbr/simple_dual``` and fixup the genimage.cfg.in to create a btrfs fs instead of ext4.

You'd maintain your external dir with it's layout knowing that you can build your image with btrfs.
How does that sound?

Thank you for taking the time to give the tool a try.
Cheers,
--  Matt

solskogen, 2025-02-07:
I think that way is a bit over engineered. A simple ROOT_FS= would be a lot easier (and ofc let default be ext4)

learmj, 2025-02-07:
Do you mean have 2 x genimage.cfg.in in ```mbr/simple_dual``` and a config variable which could be set to use the one that creates a btrfs root? That's totally fine and I can add that support if it would work for you?

I wouldn't want the in-tree layouts getting too unwieldy with various options since doing so kind of detracts from the usage ethos behind the tool, but there is definitely a case to be made for having a reasonable set of customisation options particular to the layout. More complex layouts will likely need customisation options anyway.

learmj, 2025-02-07:
I don't follow. I wouldn't have a global switch for 'ext4 vs btrfs' because it wouldn't make sense in all cases (eg squashfs layout, image that only output an initramfs, etc).

solskogen, 2025-02-07:
Ah, I see. Then your last suggestion is prefered.

commit d4d5dfe7 (2025-02-11): image: Add BTRFS support

Add support to mbr/simple_dual so the root filesystem may be created
as type BTRFS or Ext4, with the latter being the default.

Add a new configuration variable 'image_rootfs_type' specific to the
mbr/simple_dual layout. This can be passed via config file or options.
Update layout documentation with details.
Use the pre-build hook to verify image_rootfs_type value.
Add btrfs-progs in the list of dependencies.

The value of image_rootfs_type will select the input genimage config to
be used, as well as mount options in fstab.

Remove rootfstype=ext4 from the templated c [... cut]