raspberrypi/rpi-image-gen #15: Webkiosk

procount, 2025-02-10:
1. The build instructions for webkiosk have the incorrect example folder - it should be webkiosk, not kiosk.
2. When built and run, my mouse cursor was not visible on the browser, so I couldn't click the "Am I Human" option. It timed out eventually and then I realised my mouse was actually working as it went over some buttons and their highlighting changed, but it was still not visible. Probably just a webkiosk configuration setting somewhere and it is set up for touchscreen....?

learmj, 2025-02-10:
Thanks. I'll fix up the readme (it used to be called 'kiosk' but then I renamed it to 'webkiosk'). I've seen that same mouse behaviour but from what I can ascertain, some people want the cursor present in a kiosk (eg so they can navigate) and others don't (eg they may want to website to provide changing content). The example was never intended to enforce particular behaviour in this regard. I've seen the mouse cursor appear sometimes if the kiosk is booted with a keyboard connected then removed. It seems totally random.

Regardless, I think this is likely to be a Wayland related thing rather than anything specific to hardware or drivers. The current version of Cage used for the example is old. I plan to test a newer version.

learmj, 2025-02-11:
I corrected the readme in https://github.com/raspberrypi/rpi-image-gen/commit/52f95c3e911f2b85c87b7e6f23e3160fcf2ae05c. I don't think it's worthwhile looking into the cursor thing, especially as upstream haven't yet settled on behaviour or requirements. Cage #299 seems to be the latest discussion I can find on the topic.

closed 2025-02-11 by learmj

procount, 2025-02-11:
Not a problem. I don't intend to use it, just testing.