raspberrypi/rpi-image-gen #24: Build on a build server

knachte, 2025-03-04:
I've tried several times the last few months to get this to build on a pc, as from experience with pi-gen i've seen that build times can be multiple times slower on a pi. But i've never been able to get it to produce something. Are there any plans to get cross building up and running? Currently trying this on ubuntu 24.4 on  wsl on my local machine, but if this would become a path that we would follow, the security team would also force me to produce the builds on our build servers.

learmj, 2025-03-04:
Hi @knachte Thanks for your report.
I've just cloned rpi-image-gen repo on my x86_64 ubuntu 24.04 workstation and successfully built an image with it.
qemu and binfmt aren't currently listed in the dependencies because I've never tested this and so cannot declare 'cross build support'. It's literally the first time I've tried it. My ubuntu machine already has the necessary packages and sources installed, so mmdebstrap reported...

> I: arm64 cannot be executed natively, but transparently using qemu-user binfmt emulation

...and carried on quite happily (but slowly due to the emulation).

The only thing I needed to do was to specify a key dir for APT to use so it could authenticate the repositories specified in the layers pulled in by the profile, otherwise you'll see several things like:

>  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
> Err:2 http://deb.debian.org/debian bookworm InRelease
> 

Doing that is easy:
1. tar up all your keys from a Raspberry Pi device under ```/usr/share/keyrings/``` and also within ```rpi-image-gen/keys``` from a cloned repo (you'll only need raspberrypi-archive-stable.gpg.key unless you're building a 32-bit userland system from raspian).
2. On your x86 host, unpack this tarball somewhere.
3. cd rpi-image-gen and invoke:
```IGconf_sys_apt_keydir=/path/to/unpacked/key-tarball/contents ./build.sh```

Now that you've told rpi-image-gen what keys to use, it will instruct APT to use those when creating the chroot, therefore allowing it to [now] fetch and authenticate all the packages it needs. If your build host has qemu/binfmt installed and also genimage (which isn't in Ubuntu for x86 so you'll have to build it from src and install it manually), the build will complete. I successfully booted the image generated by my 'cross' build on my pi5 and it's fine.

What problems did you encounter? Hopefully the above is enough to demonstrate that rpi-image-gen can be used in a cross build scenario.

> as from experience with pi-gen i've seen that build times can be multiple times slower on a pi. 

If you try rpi-image-gen on a pi, I think you should find it faster than pi-gen. It leverages a different tool under the hood.

knachte, 2025-03-06:
Thanks, i'll give this a try after i'm back from embedded world!

jonnymacs, 2025-03-07:
@knachte I have a Dockerfile that I'm using to bootstrap a Debian system and build images with this utility. I'm planning to publish it somewhere - fairly soon. It's working well, I'm on a Mac with an M4 Pro ARM chip.

@learmj let me know if this is of any interest to you on this project?

learmj, 2025-03-11:
Thanks for the offer @jonnymacs I was contemplating adding this feature but for now, it's in the holding pattern queue.

jonnymacs, 2025-03-13:
@learmj @knachte here's the Dockerfile I'm using.

Note I add the user to the sudo group to support a post-build step that requires sudo privileges in the example (autologin) 

Note in the Readme - on an x86 chip there is a step to mount binfmt_misc in the container. I regret I don't know exactly why this was necessary and it's been a few months since I stopped using an intel chip. This can probably be added to the Dockerfile with a check for the architecture. 

https://github.com/jonnymacs/rpi-image-gen-example/blob/main/Dockerfile

learmj, 2025-03-14:
Thanks for providing the visibility of your script @jonnymacs 

> Note I add the user to the sudo group to support a post-build step that requires sudo privileges in the example (autologin)

I have wondered in the past whether I should use ```podman unshare``` when invoking the hooks. It would cover these cases and remove the need for sudo.

learmj, 2025-04-02:
@knachte do you have a working build now?

knachte, 2025-04-02:
@learmj not yet unfortunately, our new server came in a bit later than expected, we are still setting it up and will first need to set up the builds for the projects that are in active production first, but after that i'm giving it a try. I'll close this for now , i'll reopen if needed in a few weeks.

closed 2025-04-02 by knachte