raspberrypi/rpi-image-gen #29: Error with btrfs build option

LorandMarton, 2025-03-07:
# Description

Attempting to build an image with `image_rootfs_type=btrfs` results in a genimage error stating:
`/home/pi/rpi-image-gen/work/testimage/artefacts/genimage.cfg:37: no such option 'btrfs'`

# Steps to Reproduce

Use the setoptions example as the base configuration.

Add the following option: `image_rootfs_type=btrfs`

# Build process error

Full error log::

<details> <summary>Click to expand full error log</summary> 

[12:29:12] /home/pi/rpi-image-gen/device/pi5[post-build.sh] /home/pi/rpi-image-gen/work/testimage/rootfs
[12:29:12] /home/pi/rpi-image-gen/image/mbr/simple_dual[pre-image.sh] /home/pi/rpi-image-gen/work/testimage/rootfs /home/pi/rpi-image-gen/work/testimage/artefacts
[12:29:13] /home/pi/rpi-image-gen/sbom[gen.sh] /home/pi/rpi-image-gen/work/testimage/rootfs /home/pi/rpi-image-gen/work/testimage/artefacts
[info] checking github for the current release tag
[info] fetching release script for tag='v1.20.0'
[info] checking github for the current release tag
[info] using release tag='v1.20.0' version='1.20.0' os='linux' arch='arm64'
[info] installed /home/pi/rpi-image-gen/work/testimage/host/bin/syft
[12:29:22] SBOM: syft 1.20.0 scanning /home/pi/rpi-image-gen/work/testimage/rootfs
 ✔ Indexed file system                                                                                                                                             /home/pi/rpi-image-gen/work/testimage/rootfs
 ✔ Cataloged contents                                                                                                                          4919a2773a2bfbc55586b2cfd021b8d3f3cfedf473a4f11df1ada2f8261fecfa     ├── ✔ Packages                        [215 packages]
   ├── ✔ File digests                    [9,840 files]
   ├── ✔ File metadata                   [9,840 locations]
   └── ✔ Executables                     [1,219 executables]
/home/pi/rpi-image-gen/work/testimage/artefacts/genimage.cfg:37: no such option 'btrfs'
[12:29:50] Error: [podman unshare env IGconf_device_class=pi5 IGconf_device_hostname=pigen IGconf_device_keyboard_keymap=gb IGconf_device_keyboard_layout=English (UK) IGconf_device_locale1=en_GB.UTF-8 IGconf_device_options=defaults IGconf_device_profile=bookworm/apt-min64 IGconf_device_ssh_user1=n IGconf_device_timezone=Europe/London IGconf_device_timezone_area=Europe IGconf_device_timezone_city=London IGconf_device_user1=pi IGconf_device_user1pass=pi IGconf_device_variant=none IGconf_image_boot_part_size=200% IGconf_image_compression=none IGconf_image_layout=mbr/simple_dual IGconf_image_name=testimage IGconf_image_options=defaults IGconf_image_root_part_size=300% IGconf_image_rootfs_type=btrfs IGconf_image_suffix=img IGconf_image_version=2025-03-07 IGconf_sbom_output_format=spdx-json IGconf_ssh_user1=y IGconf_sys_deploydir=/home/pi/rpi-image-gen/work/testimage/deploy IGconf_sys_outputdir=/home/pi/rpi-image-gen/work/testimage/artefacts IGconf_sys_workdir=/home/pi/rpi-image-gen/work/testimage IGDEVICE=/home/pi/rpi-image-gen/device/pi5 IGIMAGE=/home/pi/rpi-image-gen/image/mbr/simple_dual IGPROFILE=/home/pi/rpi-image-gen/profile/bookworm/apt-min64 PATH=/home/pi/rpi-image-gen/bin:/home/pi/rpi-image-gen/work/testimage/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games genimage --rootpath /home/pi/rpi-image-gen/work/testimage/rootfs --tmppath /tmp/tmp.iju52ZbU4Q --inputpath /home/pi/rpi-image-gen/work/testimage/artefacts --outputpath /home/pi/rpi-image-gen/work/testimage/artefacts --loglevel=1 --config /home/pi/rpi-image-gen/work/testimage/artefacts/genimage.cfg] (1)
Generating image...0:00:00
[12:29:50] genimage error

</details>

# Additional Information

The `genimage.cfg` file:  

<details> <summary>Click to expand full error log</summary> 
 /home/pi/rpi-image-gen/work/testimage/artefacts/genimage.cfg:

image /home/pi/rpi-image-gen/work/testimage/artefacts/testimage.img.sparse {
   android-sparse {
      image = /home/pi/rpi-image-gen/work/testimage/artefacts/testimage.img
   }
}

image /home/pi/rpi-image-gen/work/testimage/artefacts/testimage.img {
   hdimage {
      partition-table-type = "mbr"
   }

   partition boot {
      in-partition-table = true
      partition-type = 0xC
      image = boot.vfat
      bootable = true
   }

   partition root {
      in-partition-table = true
      partition-type = 0x83
      image = root.btrfs
   }
}

image boot.vfat {
   vfat {
      label = "BOOT"
      extraargs = "-s 4 -S 512 -i 8acc9b45"
   }
   size = 200%
   mountpoint = "/boot/firmware"
   exec-pre = "'/home/pi/rpi-image-gen/image/mbr/simple_dual/setup.sh' BOOT 8acc9b45 12ada3d4-bbb9-4148-8391-c316333d2213"
}

image root.btrfs {
   btrfs {
      label = "ROOT"
      extraargs = "-U 12ada3d4-bbb9-4148-8391-c316333d2213"
   }
   size = 300%
   mountpoint = "/"
   exec-pre = "'/home/pi/rpi-image-gen/image/mbr/simple_dual/setup.sh' ROOT 8acc9b45 12ada3d4-bbb9-4148-8391-c316333d2213"
}
</details>

## Question

Is additional configuration required for `btrfs` build, or should the setup be adjusted in a different way?

learmj, 2025-03-07:
Hi. Thanks for the report. Template fields are much appreciated. You need an up to date version of genimage in order to support btrfs. Further info in d4d5dfe74728b6a842b23c707a329babcc7a5bf1

LorandMarton, 2025-03-11:
Thanks for your hint - an up-to-date version of genimage works well.

I’m trying to create a Raspberry Pi OS Lite-like image using rpi-image-gen for a Raspberry Pi 4B. My assumption is that adding the same packages with a YAML layer as pi-gen’s stage2 should get me close, but I suspect it’s not that simple.

My end goal is to get the Lite version running on Btrfs.

Would you have any hints on how to properly set this up? Once I get it working, I could contribute this as an example.

learmj, 2025-03-12:
Cool. Glad it works for you.
Re: lite/stage2 - I think it probably is that simple as far as the packages go, and you could also use scripts in the YAML (or add custom bdebstrap hooks) to fixup the remaining bits and pieces like file tweaks, etc. Long term for rpi-image-gen, it might be useful if it could generate a lite image. It's something I may add in the future but it's primary use at the moment is to create custom images rather than having absolute parity with a system that pi-gen creates, although I'm sympathetic with your quest to have a lite image with btrfs :-)
Maybe looking at it another way, if you know what packages you need in your custom lite image (a lite image is pretty huge) and want to leverage the in-tree ```mbr/simple_dual layout``` with brtfs,  I would suggest:

* Create a top level external ```<dir>``` with sub-directories for profile, config and meta.
* Create ```<dir>```/config/myconfig.cfg so it has:
```
[image]
layout=mbr/simple_dual
boot_part_size=whatever
root_part_size=whatever
name=yourimagename

[device]
class=pi5
profile=yourprofile
```
There is no pi4 device directory yet but I intend to add one soon, along with others.

* Create ```<dir>```/meta/yourlayer.yaml to describe the packages etc that you want, script ops, etc (can have as many layers you like). You can create <dir>/meta/yourlayer.defaults as a shell fragment if you want the layer to reference any variables or to have them set to defaults (variables will be auto prefixed with ```IGconf_meta```). Directories are supported for layers and defaults.
* Create ```<dir>```/profile/yourprofile and in it reference the layers you want from the in-tree ```meta``` and your dir, eg yourlayer etc. One layer per line.

If you find you can't do what you need to in the YAML, create ```<dir>```/bdebstrap/customize-01 etc which will get called at chroot creation time at the customise stage (which happens after essential). The bdebstrap hooks are all run after those in the YAML (see https://github.com/raspberrypi/rpi-image-gen/blob/master/bin/runner for the order and how these are run). For example, anything you do in YAML under ```customize-hooks``` is guaranteed to have been executed before the bdebstrap hooks ```customize*``` are run.

Build with:
```./build.sh -D <dir> -c myconfig.cfg```

You can also hook up any options via ```<dir>```/options and include it above using ```-o```. Options can override any setting.
You can also pass variables in the env to ```build.sh```, eg if you wanted to override meta variable myvar used in your cusom YAML you could do:

```IGconf_meta_myvar=foo ./build.sh -D <dir> -c myconfig.cfg```

Hope that helps. You may find that you don't need everything that a lite image has.

Lastly, unlike pi-gen, rpi-image-gen doesn't support any scripts to perform on-device resize. You can find some discussion about that in #5. We're targeting using sparse images now for device provisioning. You can use the https://github.com/raspberrypi/pi-gen-micro fastboot gadget with rpiboot (https://github.com/raspberrypi/usbboot) to write a sparse image with large apparent on-disk size to the device using [fastboot](https://android.googlesource.com/platform/system/core/+/refs/heads/master/fastboot/README.md). Some rudimentary size vs time info [here](https://github.com/raspberrypi/rpi-image-gen/issues/5#issuecomment-2621807356).
Cheers,
--  Matt

LorandMarton, 2025-03-27:
Hi,

I continued my quest to have a working Btrfs-based image.

Working scenario:
I started from the `setoptions` example and made the following additions to `my.options`:

```sh
DEVICE_HOSTNAME=genc
device_user1=pi
dEvIcE_UsER1pASs=pi
IMAGE_NAME=mydevice-custom-new
image_rootfs_type=btrfs
```
With this configuration, the image builds and boots successfully.

### Failing scenario

additionally, I added `device_ssh_user1=y`

Next, I created a custom profile `rpi-image-gen/examples/setoptions/profile/v8-svelte` with:
```sh
debian/bookworm/arm64/base-apt
rpi/debian/bookworm/apt
rpi/misc-utils
rpi/base/essential
rpi/boot-firmware
rpi/arm64/linux-image-v8
rpi/user-credentials
rpi/misc-skel
sys-apps/systemd-net-min
sys-apps/fake-hwclock
slim-customisations
````

With the following meta configuration: `rpi-image-gen/examples/setoptions/meta/slim-customisations.yaml`:

```sh
---
name: example-slim
mmdebstrap:
  packages:
    - build-essential
    - wpasupplicant
    - wireless-tools
    - firmware-atheros
    - firmware-brcm80211
    - firmware-libertas
    - firmware-misc-nonfree
    - firmware-realtek
    - raspberrypi-net-mods
    - net-tools
    - avahi-daemon
    - libnss-mdns
```
And: `rpi-image-gen/examples/setoptions/config/pi5-slim.cfg`:
```sh
[device]
class=pi5
profile=v8-svelte

[image]
layout=mbr/simple_dual
```

Built with: `./build.sh -D ./examples/setoptions/ -c pi5-slim -o ./examples/setoptions/my.options`

### Problem

Flashed the image onto a USB stick. On first boot, I got the following:

```sh
BTRFS error (device sda2): device total_bytes should be at most 830472192 but found 17332296128
BTRFS error (device sda2): failed to read chunk tree: -22
mount: Invalid argument
Failed to mount /dev/sda2 as root file system.
```

Then I resized the partition and filesystem manually, after which the system reached further in the boot process but failed with:

```sh
[FAILED] Failed to mount /proc-sys-fs-binfmt_misc.mount - Arbitrary Executable File Formats File System
[FAILED] Failed to start systemd-timesyncd.service - Network Time Synchronization.
...
BTRFS warning (device sda2): csum failed root 5 ino 16530 off 0 csum 0xc97700b1 expected csum 0x6590cfda2
BTRFS error (device sda2): bdev /dev/sda2 errs: wr 0, rd 0, flush 0, corrupt 50, gen 0
BTRFS warning (device sda2): csum failed root 5 ino 19431 off 0 csum 0x22ac90c7 expected csum 0x53063f0e
....
Failed to execute /sbin/sulogin: Input/output error

```

Cheers,
Lorand

learmj, 2025-03-27:
How are you writing the image to the device? Sounds like something is going wrong at that stage rather than the build stage.

If you have btrfs progs installed on your build host you should be able to ```kpartx``` the final disk image, set up loop devices then mount the partitions within the image. If you can, you can be sure the partition and its filesystem are fine, meaning it's writing it to the device that's the problem.

LorandMarton, 2025-03-27:
I used the following commands:

`sudo dd if=/home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img of=/dev/sda bs=4M status=progress conv=fsync`

checked the `kpartx`:

`sudo kpartx -av /home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img
add map loop0p1 (254:0): 0 98304 linear 7:0 16384
device-mapper: reload ioctl on loop0p2 (254:1) failed: Invalid argument
create/reload failed on loop0p2`

`sudo fdisk -l /home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img
Disk /home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img: 840.92 MiB, 881766400 bytes, 1722200 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                                                                          Boot  Start     End Sectors  Size Id Type
/home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img1 *     16384  114687   98304   48M  c W95 FAT32 (LBA)
/home/pi/rpi-image-gen/work/mydevice-custom-new/deploy/mydevice-custom-new.img2      114688 1736703 1622016  792M 83 Linux`

learmj, 2025-03-28:
Hi. Thanks for the update. This is a btrfs specific thing (e.g. ext4 does not behave this way). It's interesting that btrfs doesn't appear to be happy working within the bounds given to it by mkfs.btrfs and needs to 'extend itself', but I wonder if there are additional options that can be passed to it via genimage ```extraargs``` to avoid this.

I don't recommend using dd to write an image to storage, particularly if you want to be able to boot from it. If you must use primitive block level tools, rpi-imager is more reliable (a) because it verifies and (b) because it syncs IO and re-reads the partition table at the end of writing. If you're writing images to storage on a Raspberry Pi device I highly recommend using [pi-gen-micro](https://github.com/raspberrypi/pi-gen-micro/tree/main)'s fastboot or usb-mass-storage gadgets with rpiboot. That way you can leverage the power of sparse images using fastboot or something like bmaptool. You can write images much quicker and reliably.

learmj, 2025-03-28:
I can see what's happened here but right now I'm not quite sure why it has happened.
I build with this ```IGconf_image_rootfs_type=btrfs IGconf_image_root_part_size=100%```

The failure is this regardless of rw or ro mount:
```
[    3.876573] BTRFS info (device mmcblk0p2): using free space tree
[    3.885775] BTRFS error (device mmcblk0p2): device total_bytes should be at most 461373440 but found 1095761920
[    3.895909] BTRFS error (device mmcblk0p2): failed to read chunk tree: -22
[    3.902925] BTRFS error (device mmcblk0p2): open_ctree failed
```
...which renders the partition unmountable forevermore. I can't mount it on a linux host and ```btrfs check``` fails.

However, the BTRFS partition image generated by the build is clean and can be mounted just fine.

The root cause of the problem appears to be a mismatch in sizes between the partition table entry containing the filesystem and the filesystem itself:
```
            "partitions": [
                {
                    "start": 16384,
                    "size": 180224,
                    "type": "c",
                    "bootable": true
                },
                {
                    "start": 196608,
                    "size": 901120,
                    "type": "83"
                }
            ]
```
```
BTRFS error (device mmcblk0p2): device total_bytes should be at most 461373440 but found 109576192
```
901120 x 512 = 461373440

...however when I mount the BTRFS partition image it's total size is reported as 109576192 which is totally wrong:
```
/dev/loop0       1095761920   468176896    549400576  47% /tmp/mnt
```
If I rebuild the exact same configuration with ```IGconf_image_rootfs_type=ext4 IGconf_image_root_part_size=100%```

```
/dev/loop0        411039744   366811136     17532928  96% /tmp/mnt
```

The total size reported is correct.

The genimage configs for ext4 and btrfs don't have any differences which could account for this so it appears to be a bug in either genimage when creating btrfs filesystems or mkfs.btrfs...

lurch, 2025-03-28:
I don't have anything to add to this conversation, but some of those huge numbers look much "neater" when converted to hexadecimal 😉  (I also inserted thousands separators below)

- 16_384 = 0x4_000
- 180_224 = 0x2C_000
- 196_608 = 0x30_000
- 901_120 = 0xDC_000
- 461_373_440 = 0x1B_800_000 = 440 MiB
- 1_095_761_920 = 0x41_500_000 = 1045 MiB
- 411_039_744 = 0x18_7FF_800 = 391.998 MiB

1095761920 bytes is 2.375 × 461373440 bytes (or to put it another way, (440 MiB  × 2) + 165 MiB = 1045 MiB).
I wondered if maybe 440 MiB was less than the the minimum btrfs partition size, but a google search suggests that the minimum btrfs partition size is 256 MiB.

LorandMarton, 2025-03-30:
@learmj, what would be the correct way to add Btrfs subvolumes using rpi-image-gen? I’ve checked and it seems that genimage doesn’t support this.

learmj, 2025-03-31:
> what would be the correct way to add Btrfs subvolumes using rpi-image-gen?

The discrete image layout + hooks make it so you have complete control over your image, so I'd suggest using them in such a way that meets your needs / purposes. subvolumes will be specific to a user/configuration so I don't plan making them part of the in-tree btrfs configuration in the current image layout that uses them.

> Is it possible to preinstall Python packages with pip?

You mean in a build generated with rpi-image-gen? I'd have thought you can run a hook.
If you want to play with fastboot, I'd recommend just installing it via pi-gen-micro (or install the it's deb pkg).

learmj, 2025-04-02:
This isn't an rpi-image-gen bug so I'm closing it. I proposed a change upstream for genimage which at least catches the case where an fs image is larger than the partition it's intended to fill.

closed 2025-04-02 by learmj