raspberrypi/rpi-image-gen #36: How to run this in docker container in x64 systems.

gxanshu, 2025-03-23:
i tried to create a docker image that is base on debian bookworm and i used `--platform=linux/arm64` 
here is my docker file

```Dockerfile
FROM --platform=linux/arm64 debian:bookworm AS base

RUN apt-get update && apt-get install --no-install-recommends -y \
    build-essential \
    curl \
    git \
    ca-certificates \
    sudo \
    gpg \
    gpg-agent \
    dirmngr \
    slirp4netns \
    quilt \
    parted \
    qemu-user-static \
    binfmt-support \
    zerofree \
    libcap2-bin \
    libarchive-tools \
    xxd \
    file \
    kmod \
    bc \
    pigz \
    arch-test \
    && rm -rf /var/lib/apt/lists/*

RUN curl -fsSL https://archive.raspberrypi.com/debian/raspberrypi.gpg.key \
    | gpg --dearmor > /usr/share/keyrings/raspberrypi-archive-keyring.gpg

RUN git clone https://github.com/raspberrypi/rpi-image-gen.git

RUN /bin/bash -c 'apt-get update && rpi-image-gen/install_deps.sh'

ENV USER imagegen
RUN useradd -u 4000 -ms /bin/bash "$USER" && echo "${USER}:${USER}" | chpasswd && adduser ${USER} sudo # only add to sudo if build scripts require it
USER ${USER}
WORKDIR /home/${USER}

RUN /bin/bash -c 'cp -r /rpi-image-gen ~/'
```

when i tried to run ./build.sh then this is what i get

```
[02:31:24] /home/imagegen/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
cannot clone: Invalid argument
Error: cannot re-exec process
[02:31:24] Error: [podman unshare bdebstrap --config /home/imagegen/rpi-image-gen/meta/debian/bookworm/arm64/base-apt.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/debian/bookworm/apt.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/misc-utils.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/base/essential.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/boot-firmware.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/arm64/linux-image-v8.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/user-credentials.yaml --config /home/imagegen/rpi-image-gen/meta/rpi/misc-skel.yaml --config /home/imagegen/rpi-image-gen/meta/sys-apps/systemd-net-min.yaml --config /home/imagegen/rpi-image-gen/meta/sys-apps/fake-hwclock.yaml --env IGconf_device_class=pi5 --env IGconf_device_hostname=raspberrypi --env IGconf_device_keyboard_keymap=gb --env IGconf_device_keyboard_layout=English (UK) --env IGconf_device_locale1=en_GB.UTF-8 --env IGconf_device_profile=bookworm/apt-min64 --env IGconf_device_ssh_user1=n --env IGconf_device_storage_type=sd --env IGconf_device_timezone=Europe/London --env IGconf_device_timezone_area=Europe --env IGconf_device_timezone_city=London --env IGconf_device_user1=pi --env IGconf_device_variant=none --env IGconf_image_boot_part_size=200% --env IGconf_image_compression=none --env IGconf_image_layout=mbr/simple_dual --env IGconf_image_name=deb12-arm64-min --env IGconf_image_root_part_size=300% --env IGconf_image_rootfs_type=ext4 --env IGconf_image_suffix=img --env IGconf_image_version=2025-03-23 --env IGconf_sbom_output_format=spdx-json --aptopt APT::Get::Purge true --aptopt Dir::Etc::TrustedParts /home/imagegen/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_apt_keydir=/home/imagegen/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_deploydir=/home/imagegen/rpi-image-gen/work/deb12-arm64-min/deploy --env IGconf_sys_outputdir=/home/imagegen/rpi-image-gen/work/deb12-arm64-min/artefacts --env IGconf_sys_target=/home/imagegen/rpi-image-gen/work/deb12-arm64-min/rootfs --env IGconf_sys_workdir=/home/imagegen/rpi-image-gen/work/deb12-arm64-min --env IGTOP=/home/imagegen/rpi-image-gen --env META_HOOKS=/home/imagegen/rpi-image-gen/meta-hooks --env RPI_TEMPLATES=/home/imagegen/rpi-image-gen/templates/rpi --env IGDEVICE=/home/imagegen/rpi-image-gen/device/pi5 --env IGIMAGE=/home/imagegen/rpi-image-gen/image/mbr/simple_dual --env IGPROFILE=/home/imagegen/rpi-image-gen/profile/bookworm/apt-min64 --env PATH=/home/imagegen/rpi-image-gen/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin --force --name deb12-arm64-min --hostname raspberrypi --output /home/imagegen/rpi-image-gen/work/deb12-arm64-min/artefacts --target /home/imagegen/rpi-image-gen/work/deb12-arm64-min/rootfs --setup-hook bin/runner setup "$@" --essential-hook bin/runner essential "$@" --customize-hook bin/runner customize "$@" --cleanup-hook bin/runner cleanup "$@"] (125)
```

learmj, 2025-03-24:
Hi. Thanks for the report. At the moment, there is no claimed support for building on a machine which has a different arch to the output target. I did try it for one of the purposes of the following tickets and it worked, but it wasn't in a Docker container.
https://github.com/raspberrypi/rpi-image-gen/issues/24
https://github.com/raspberrypi/rpi-image-gen/issues/26
Some information in one or both these tickets might be of interest to you. Not sure if you've read them?

What mode did mmbedstrap report it was running in for your case? I suspect there was more debug info provided and you just didn't include it here.

jonnymacs, 2025-03-24:
@gxanshu that code looks familiar, (you probably got it from my example repo) with a few changes I can see - you added

`--platform=linux/arm64`

to the base layer import.

When I use your exact Dockerfile to build an image, I am able to build a rpi image using the ./build.sh script on my M4 Mac with Docker Desktop Version 4.39.0 (184744).

Some things to try

Make sure you run the container in privileged mode. When I don't run it in privileged mode I see a slightly different error but at the same point in the build script - invoking `podman unshare bdebstrap` after pre-build.sh

Digging a little - when running podman inside a docker container you need to ensure the host is able to allow podman to do what it want.

Grok or Copilot might also help you debug further. Prompting Grok with your error, it provided some steps to help debug, starting with ensuring you run with `--privileged`

> Podman relies heavily on user namespaces and system calls like clone() to create containers, especially in rootless mode 
> (which podman unshare facilitates). Docker, by default, restricts these capabilities unless you explicitly enable them.

jonnymacs, 2025-03-25:
@learmj agreed. It is not an issue with this tool.

@gxanshu if you want to open an issue against [my example repo](https://github.com/jonnymacs/rpi-image-gen-example/issues) I wouldn't mind using that space to discuss so we can record the solution.

jonnymacs, 2025-03-25:
@gxanshu I've gone ahead and [opened one](https://github.com/jonnymacs/rpi-image-gen-example/issues/1)

@learmj if you want to close this here.

learmj, 2025-03-26:
Thanks @jonnymacs I'll keep an eye on the new ticket. I appreciate you taking it on. If running in Docker starts to cause lots of issues because of the usage of podman, I may add some sort of troubleshooting section to the rpi-image-gen readme.

closed 2025-03-26 by learmj