raspberrypi/rpi-image-gen #37: Debian 12 support

corv89, 2025-03-25:
To run `./build.sh` on a stock Debian Bookworm installation, I needed to:

1. Add the raspberry pi gpg key to the keyring:

```bash
cp keydir/raspberrypi-archive-stable.gpg.key /etc/apt/keyrings/
```

2. Add /sbin to path (for dosfstools)
```bash
export PATH=$PATH:/sbin:/usr/sbin
```

It would be nice if these steps could be automated, since Raspberry Pi OS can't be run easily on Arm Macs.

Other than that, this is a great tool to make custom images quickly, thanks to everyone involved!

learmj, 2025-03-25:
Hi @corv89 Thanks for the report and the feedback.

You're right that you need to copy the key needed to authenticate rpi APT packages if it's not in a location that APT knows about by default when creating the chroot. This is covered in the documentation here:
https://github.com/raspberrypi/rpi-image-gen/blob/master/sys-build.defaults#L16

Regarding, usage of host utilities, Debian seems to not make /sbin available to a regular user by default. This is Debian's choice but is unlike some other distributions, eg Gentoo. Also, Raspberry Pi OS (which is the defacto support distribution rpi-image-gen runs on) also puts ```/usr/sbin``` in a regular user's PATH by default.

This might make for interesting reading:
[https://lists.debian.org/debian-devel/1999/12/msg00943.html](url)

corv89, 2025-03-25:
Thanks for the quick response.

For reference, these keys work on Ubuntu 24.04.2 LTS:

```bash
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0E98404D386FA1D9 6ED0E7B82643E131 F8D2585B8783D481
```

I understand `rpi-image-gen` is targeting Raspberry Pi OS and that makes perfect sense. Unfortunately running RPi OS would require Qemu incantations for those like me who don't develop on a Raspberry.

learmj, 2025-03-25:
I do see the 'auto appending to PATH' as a host configuration thing, and something outside the bounds of what rpi-image-gen should do, IMHO. If things like ```mkdosfs``` aren't in your PATH then the dependency checker should have failed before the build even starts. This would alert that the program wasn't found. Did that not fail in your case?

You'd have the same problem if you tried to run mkfs.foo outside of rpi-image-gen, which kind of reinforces the point that it's not something that rpi-image-gen should workaround transparently.

corv89, 2025-03-25:
I respect projects with a strong focus and you're right that it's Debian that's out of line here.

For what it's worth, I can build conveniently with a Dockerfile; the one provided by @gxanshu was helpful in this regard.

I'd be happy to reference my solution here for posterity and would then close the issue. Glad we could have a civil discussion about it.

gxanshu, 2025-03-25:
@corv89 please share your dockerfile and processor architecture if it works for you.

corv89, 2025-03-25:
> [@corv89](https://github.com/corv89) please share your dockerfile and processor architecture if it works for you.

Since I'm using a arm64 host to build, I suggest you try using Github Actions arm64 runners, otherwise there's also very cheap arm64 VPS at Hetzner or Scaleway for example.

corv89, 2025-03-25:
Here's my Dockerfile:

```dockerfile
FROM debian:bookworm AS base

# Install minimal dependencies in a single layer
RUN apt-get update && apt-get install --no-install-recommends -y \
    git ca-certificates gpg curl \
    && rm -rf /var/lib/apt/lists/*

# Set up work directory
WORKDIR /rpi-image-gen
COPY . /rpi-image-gen/

# Install build dependencies
RUN apt-get update && /rpi-image-gen/install_deps.sh && rm -rf /var/lib/apt/lists/*

# Create non-root user
ENV USER=builder
RUN useradd -u 4000 -ms /bin/bash "$USER" && \
    chown -R ${USER}:${USER} /rpi-image-gen

# Switch to non-root user
USER ${USER}
WORKDIR /home/${USER}

# Copy repo to user's home directory
RUN cp -r /rpi-image-gen ~/

# Set working directory to repo copy
WORKDIR /home/builder/rpi-image-gen

ENTRYPOINT ["./build.sh"]
CMD []
```

It's built with `docker build -t rpi-image-gen -f Dockerfile .`
and run with `docker run -it --rm --privileged rpi-image-gen`

closed 2025-03-25 by corv89

learmj, 2025-03-26:
Hi @corv89 

> Glad we could have a civil discussion about it.

Absolutely :-) I really appreciate any feedback on the tool. I want to make it work for as many users as possible and for it to be a solid choice for developers.
If you have feature requests or anything which you think would make your workflow better/easier if rpi-image-gen 'did it that way', I'd definitely consider it.

Thanks for taking the time to try it.