raspberrypi/rpi-image-gen #38: x86 cross building update-initramfs speed issue

jove1, 2025-03-25:
I run rpi-image-gen in docker container on x86 machine (with binfmt_misc emulation)

Dockerfile:
```
FROM debian:bookworm-slim
RUN apt-get update && apt install --no-install-recommends -y ca-certificates git sudo zip dosfstools rsync curl genimage mtools mmdebstrap bdebstrap podman crudini zstd pv uidmap python-is-python3 dbus-user-session btrfs-progs dctrl-tools uuid-runtime curl arch-test && rm -rf /var/lib/apt/lists/*
RUN curl -fsSL https://archive.raspberrypi.com/debian/raspberrypi.gpg.key | gpg --dearmor > /usr/share/keyrings/raspberrypi-archive-keyring.gpg
ENV USER user
RUN useradd -m $USER && adduser $USER sudo  && echo '%sudo ALL=(ALL) NOPASSWD:ALL' >> /etc/sudoers
USER $USER
WORKDIR /home/$USER/rpi-image-gen
```
build command:
```
time docker run -it --rm --privileged \
        -v `pwd`/rpi-image-gen:/home/user/rpi-image-gen \
        -v `pwd`/work:/home/user/rpi-image-gen/work \
        -v `pwd`/conf:/home/user/rpi-image-gen/conf \
        rpi-image-gen sh -c "sudo mount -t binfmt_misc none /proc/sys/fs/binfmt_misc; ./build.sh" | tee build.log
```

`update-initramfs` runs total 3 times making up significant part of total build time (9min)

```
...
Setting up initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: deferring update (trigger activated)
...
Setting up linux-image-6.6.74+rpt-rpi-v8 (1:6.6.74-1+rpt1) ...
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-v8
...
Processing triggers for initramfs-tools (0.142+rpt3+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-v8
...
I: running --customize-hook in shell: sh -c 'sed -i 's/^update_initramfs=.*/update_initramfs=no/' $1/etc/initramfs-tools/update-initramfs.conf' exec /home/user/rpi-image-gen/work/deb12-arm64-min/rootfs
...
selecting initramfs modules
update-initramfs: Generating /boot/initrd.img-6.6.74+rpt-rpi-v8
...
```

Two times after initramfs-tools and kernel installation (why 2 times? debian bug?).
Third invocation is from `./scripts/bdebstrap/customize10-initramfs`.
Hook in `meta/rpi/arm64/linux-image-v8.yaml` setting `update_initramfs=no` is called too late to have any influence.

Uncompressing all modules on emulated architecture is very slow.

I tried to select only required modules (though I didn't check I got them all).
`./build.sh -D conf`

conf/bdebstrap/customize05-select-modules:
```
#!/bin/sh

echo "selecting initramfs modules"

cat > $1/etc/initramfs-tools/modules <<END
sdhci_iproc
mmc_block
ext4
END

sed -i \
  -e "s|MODULES=\([^ ]*\)|MODULES=list|" \
  -e "s|COMPRESSION=\([^ ]*\)|COMPRESSION=gzip|" \
  $1/etc/initramfs-tools/initramfs.conf
```

Unfortunately this only affects last `update-initramfs` invocation reducing the build time by 1min 20s.

How to avoid 2 invocations before custom config can take effect?
Why is `update_initramfs=no` hook called late?
If there is already customization of initramfs, wouldn't it be possible to build it on the host?

learmj, 2025-03-26:
Hi. Thanks for the report. That is very inconvenient indeed.

The order of packages is important here because if initramfs-tools is pulled in before the ```customize``` ops in ```meta/rpi/arm64/linux-image-*``` sets ```update_initramfs=no``` in ```/etc/initramfs-tools/update-initramfs.conf```, any call to ```update_initramfs``` will rebuild.

rpi-image-gen shouldn't assume that ```initramfs-tools``` will be installed so it can't (for example) create an early ```initramfs-tools``` with ```update_initramfs=no```.

I've been aware of this for a while and expecting the problem to surface in some custom builds, but as you're running the default build target I'd expect the initramfs to be built once (that's what I've always seen).

> Why is update_initramfs=no hook called late?

This is done by ```meta/rpi/arm64/linux-image-*``` and the order of which the YAML layers are aggregated together dictates when this hook is run. This is not ideal for this package. We almost need a ```postinst``` hook for the ```initramfs-tools``` package itself so that it can be pulled in any time without the user incurring unnecessary rebuilds of the initramfs images.

Essentially, ```update_initramfs=no``` needs to be set as early as possible **if** ```initramfs-tools``` is installed (either by a layer or as a pkg dep).

```scripts/bdebstrap/customize10-initramfs``` runs after all YAML aggregation and device + image bdebstrap hooks are run, and it will trigger the initramfs image build.

This is something I will look into. Suggestions are welcome.

jove1, 2025-03-26:
First double invocation seems to be debian bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1099136

jove1, 2025-03-28:
This seems to mitigate the issue, care must be taken to include needed modules in the list
(will check later what is actually needed when I test on real hw).

Dropping `update_initramfs=no` into  `/etc/initramfs-tools/update-initramfs.conf` from essential-hook
triggered interactive dpkg question about old config file and did not stop mkinitramfs run after kernel install.

Initramfs is still built 3 times but is much smaller so compression is faster (and no module decompression takes place).
Build time is now 5:13

```
diff --git a/meta/rpi/arm64/linux-image-v8.yaml b/meta/rpi/arm64/linux-image-v8.yaml
index 1e783b0..93bc263 100644
--- a/meta/rpi/arm64/linux-image-v8.yaml
+++ b/meta/rpi/arm64/linux-image-v8.yaml
@@ -6,5 +6,7 @@ mmdebstrap:
   packages:
     - initramfs-tools
     - linux-image-rpi-v8
+  essential-hooks:
+    - mkdir -p $1/etc/initramfs-tools/conf.d && printf "MODULES=list\nCOMPRESS=gzip\n" > $1/etc/initramfs-tools/conf.d/limit_modules
   customize-hooks:
     - sed -i 's/^update_initramfs=.*/update_initramfs=no/' $1/etc/initramfs-tools/update-initramfs.conf
```

learmj, 2025-03-28:
> Dropping update_initramfs=no into /etc/initramfs-tools/update-initramfs.conf from essential-hook
> triggered interactive dpkg question about old config file and did not stop mkinitramfs run after kernel install.

I guess that's why you see it built more than once then. I've looked at this today and tried to reproduce but I cannot. I only ever see it built once. I've run the tool on an x86 box with binfmt emulation. It's slow but it only ever builds initramfs once.
Having ```update_initramfs=no``` in  ```/etc/initramfs-tools/update-initramfs.conf``` should stop all (re)builds.

> triggered interactive dpkg question about old config file

This is odd, too. Sounds like there is some underlying issue causing this.

I've barely done any work or verification running rpi-image-gen in a Docker container let along on a machine with a different arch. I know it builds on my Ubuntu 24.04 x86 box. That has been the extent of my 'testing'.

If you chroot into your build rootfs, make sure ```update_initramfs=no``` is set in  ```/etc/initramfs-tools/update-initramfs.conf``` and invoke ```sudo update-initramfs -uv```, what happens?

jove1, 2025-03-28:
`update-initramfs -u` says `update-initramfs: Not updating initramfs.`

First call is from `/etc/kernel/postinst.d/initramfs-tools` where the call is `update-initramfs -c -k 6.6.74+rpt-rpi-v8` (unaffected by `update_initramfs=no`). Other calls are `-u` and correctly turned off by `update_initramfs=no`. However, customize-hook is called after first `-u` invocation (due to initramfs-tools installation). When creating `/etc/initramfs-tools/update-initramfs.conf` in essential-hook (before initramfs-tools install) dpkg then asks interactively about different versions of that file.

learmj, 2025-03-28:
Got it. Thanks. I have some thoughts on how this could be addressed. I'll cc you on the PR when I submit it.

closed 2025-06-20 by learmj