raspberrypi/rpi-image-gen #40: Specific class for pi3(b) ?

daybyter, 2025-04-01:
Hello!

Thanks a lot for your great tool. I am trying to create an image for an infoterminal, that currently uses a pi3b . AFAIK, I could use the config for pi4 with minor performance disadvantage. 

Are there plans to introduce specific settings for the pi3 , or do you consider them as unnecessary?

Thanks in advance

learmj, 2025-04-01:
Hi @daybyter   Thanks for your feedback. I appreciate you giving rpi-image-gen a try.

To be honest, at the moment, the device class directories do very little. They only set build defaults applicable to the device class the build is intended to target (which right now, is nothing really). Long term, I see those directories defining things like the kernel and firmware packages to install, and basically ensuring that only assets installed from those packages which are applicable to the device are retained in the filesystem. For example, there is no point in firmware files installed from ```raspi-firmware``` specific to a Pi4 being present in a filesystem intended to run on a Pi5. This is the opposite approach taken by RPi OS which, for obvious reasons, is intended to be able to run on any 64-bit capable device (for example). rpi-image-gen is intended to be able to create you an image specific to the device you want it to run on.

Today, you basically choose what kernel you want to run and this is contained in your top level profile, eg
https://github.com/raspberrypi/rpi-image-gen/blob/master/profile/bookworm/apt-min64#L8

In the future, for packages that are associated to the underlying device hardware (eg kernel, boot firmware, wlan / BT firmware etc) the in-tree device class directories in rpi-image-gen will dictate what packages are installed, and what files will be removed and retained from them.

For the time being, I'd have thought you can create an image using the ```config/generic64*.cfg``` config file (if you want a 64-bit base system) or the ```raspbian-bookworm-v7-apt.cfg``` (if you want 32-bit armhf native base system).

I haven't added any specific ```[pi3]``` filters in the generic config.txt that gets installed:
https://github.com/raspberrypi/rpi-image-gen/blob/master/templates/rpi/boot-firmware/config.txt

If there are sane defaults you'd like in there for pi3 (eg UART), let me know and I'll add them.

LorandMarton, 2025-04-11:
Hi!

I tried to create a 32-bit armhf native base system but I was unsuccessful, The system couldn't boot.

Here is my simple config:

my.options
```
DEVICE_HOSTNAME=gen
device_user1=pi
dEvIcE_UsER1pASs=pi
IMAGE_NAME=mydevice
```

myconfig.cfg
```
[image]
layout=mbr/simple_dual
name=raspian-deb12-min

[device]
class=pi4
profile=raspbian/bookworm-v7-apt-min
```

I also tried using a profile with armhf support (from the custom_layers example):

```
debian/bookworm/arm64/base-apt-multi
rpi/debian/bookworm/apt
rpi/misc-utils
rpi/base/essential
rpi/boot-firmware
rpi/arm64/linux-image-2712
rpi/user-credentials
rpi/misc-skel
sys-apps/systemd-net-min
sys-apps/fake-hwclock
```

Error:
```
[    0.752170] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    0.752410] fe201000.serial: ttyAMA1 at MMIO 0xfe201000 (irq = 42, base_baud = 0) is a PL011 rev3
[    0.752473] serial serial0: tty port ttyAMA1 registered
[    0.752635] 3f215040.serial: ttyS0 at MMIO 0xfe215040.serial: there is not valid maps for state default
[    0.752663] printk: legacy console ttyS0 disabled
[    0.752669] fe215040.serial: ttyS0 at MMIO 0xfe215040 (irq = 43, base_baud = 62500000) is a 16550
[    0.752731] printk: legacy console ttyS0 enabled
[    1.110488] bcm2835-auxwdt bcm2835-auxwdt: Broadcom BCM2835 watchdog timer
[    1.110920] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    1.111330] mmc-bcm2835 fe300000.mmc: mmc_debug:0 mmc_debug2:0
[    1.111341] mmc-bcm2835 fe300000.mmc: DMA channel allocated
[    1.812061] cgroup: cgroup2: unknown option "nsdelegate"
[    1.918447] of_cfs_init: OK
[    1.918586] clk: Disabling unused clocks
[    1.921741] PM: genpd: Disabling unused power domains
[    1.974971] Freeing unused kernel image (initmem) memory: 1024K
[    1.954601] Run /init as init process

Loading, please wait...
Starting systemd-udevd version 252.36-1~deb12u1+rpi1
[    2.033741] mmc1: new high speed SDIO card at address 0001
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... done.
Begin: Waiting for root file system ... Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
Begin: Running /scripts/local-block ... done.
[   32.482665] vcc-sd: disabling
Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT! UUID=549028aaf-0f61-4126-b0ff-83dd401b8647 does not exist. Dropping to a shell!
(initramfs)
````

I also got the following warnings during the build, but these are also present in the case of a working config
```
I: automatically chosen format: directory
W: Download is performed unsandboxed as root as file /home/pi/rpi-image-gen/work/mydevice/rootfs/var/lib/apt/lists/partial couldn't be accessed by user _apt
I: running --setup-hook in shell: sh -c 'bin/runner setup "$@"' exec /home/pi/rpi-image-gen/work/mydevice/rootfs
I: running apt-get update...
done
I: downloading packages with apt...
done
I: extracting archives...
done
mount: /home/pi/rpi-image-gen/work/mydevice/rootfs/sys: permission denied.
       dmesg(1) may have more information after failed mount system call.
mount: /home/pi/rpi-image-gen/work/mydevice/rootfs/proc: permission denied.
       dmesg(1) may have more information after failed mount system call.
mount: /home/pi/rpi-image-gen/work/mydevice/rootfs/proc: permission denied.
       dmesg(1) may have more information after failed mount system call.
W: since mounting /proc normally failed, /proc is now bind-mounted instead
```

@daybyter Did you succeed? 

@learmj

learmj, 2025-04-11:
Hi @LorandMarton 

> UUID=549028aaf-0f61-4126-b0ff-83dd401b8647 does not exist

The initramfs was not able to locate the root device by UUID via udev.

Layout simple/dual creates a file called fs_uuids which is used to set the filesystem UUID in each image created.
Is this UUID listed in there? It should match what's in the artefact dir genimage.cfg and also the root filesystem image itself (eg root.ext4). You can dump that out like this:
```
dumpe2fs /path/to/artefacts/root.ext4 | grep UUID
```
Lastly, this should match what's on the cmdline provided to the kernel by the bootloader via cmdline.txt. You can inspect this by either one of the following ways:

1. kpartx the image, mount the loop device associated to the  boot partition
2. mount artefacts/boot.vfat directly

cat ```/mnt/foo/cmdline.txt```

All the UUIDs should match.

LorandMarton, 2025-04-11:
Thank you,

I checked these as you suggested and the UUID matches.

however, so far I haven't tested this image with `kpartx` and with the config above I got the following error:

```
device-mapper: reload ioctl on loop0p2 (254:1) failed: Invalid argument
create/reload failed on loop0p2
```

then I added, but same error:

```
boot_part_size=200%
root_part_size=300%
```

with `root_part_size=2GB` kpartx worked. But still the same boot issue.

learmj, 2025-04-11:
Yup... I guess that's a BTRFS image?
Depending on your ```IG_conf_image_rootfs_size``` the image will not be usable.
See https://github.com/pengutronix/genimage/issues/287 and your https://github.com/raspberrypi/rpi-image-gen/issues/29

learmj, 2025-04-11:
Hi @daybyter 

> What I am currently struggling with is to run rpi-image-gen in a qemu image,

Other folks have this working. There are a few tickets to look at for reference, eg https://github.com/raspberrypi/rpi-image-gen/issues/36

LorandMarton, 2025-04-11:
> Yup... I guess that's a BTRFS image? Depending on your `IG_conf_image_rootfs_size` the image will not be usable. See [pengutronix/genimage#287](https://github.com/pengutronix/genimage/issues/287) and your [#29](https://github.com/raspberrypi/rpi-image-gen/issues/29)

I first tried using btrfs, then created the simple configuration above without btrfs. I also found this quite odd

learmj, 2025-04-11:
Trying ```root=/dev/mmcblk0p2``` directly (or whatever it is) rather than via UUID. Just hack that into ```image/mbr/simple_dual/setup.sh``` here:

```
   BOOT)
      sed -i "s|root=\([^ ]*\)|root=UUID=${ROOTUUID}|" $IMAGEMOUNTPATH/cmdline.txt
      ;;

```
You'll be able to rebuild with just ```-i``` as it will simply recreate the image rather than building everything from scratch.

LorandMarton, 2025-04-12:
I tried this but the issue unfortunately persists.

daybyter, 2025-04-12:
@LorandMarton : just yesterday late night I had the same issue with the local-block and waiting for the device failed. Since I thought, that my image should work, I just wrote my image again and now it booted. I guess this was a problem of my SD card, where some memory cells might have failed. Did you try a different SD card with the same image?

LorandMarton, 2025-04-13:
@daybyter I also tried with a different SD card, but still no luck.

daybyter, 2025-04-13:
What irritates me is this message, that you install software in your bootstrap file as the wrong user, it seems? 

Did you take a look at the bootstrap file of the kiosk example? That helped me a lot

LorandMarton, 2025-04-14:
@daybyter 

Here is my simple config, modified the setoptions example:

my.options
```
DEVICE_HOSTNAME=gen
device_user1=pi
dEvIcE_UsER1pASs=pi
IMAGE_NAME=mydevice
```

config/my-raspbian-bookworm-v7-apt.cfg
```
[image]
layout=mbr/simple_dual
boot_part_size=200%
root_part_size=5GB
name=raspian-deb12-min

[device]
class=pi4
profile=raspbian/bookworm-v7-apt-min
```

My build command: 

`./build.sh -D ./examples/setoptions/ -c my-raspbian-bookworm-v7-apt -o ./examples/setoptions/my.options`

and the first few lines of the build output:
```sh
[15:15:48] Reading /home/pi/rpi-image-gen/examples/setoptions/config/my-raspbian-bookworm-v7-apt.cfg with options [/home/pi/rpi-image-gen/examples/setoptions/my.options]
[15:15:49] IGDEVICE /home/pi/rpi-image-gen/device/pi4
[15:15:49] IGIMAGE /home/pi/rpi-image-gen/image/mbr/simple_dual
[15:15:49] IGPROFILE /home/pi/rpi-image-gen/profile/raspbian/bookworm-v7-apt-min
[15:15:50] Load profile /home/pi/rpi-image-gen/profile/raspbian/bookworm-v7-apt-min
[15:15:50] Load layer [main] raspbian/bookworm/base-apt
[15:15:50] Load layer [main] rpi/debian/bookworm/apt
[15:15:50] Load layer [main] rpi/misc-utils
[15:15:50] Load layer [main] rpi/base/essential
[15:15:50] Load layer [main] rpi/boot-firmware
[15:15:50] Load layer [main] rpi/armhf/linux-image-v7
[15:15:50] Load layer [main] rpi/user-credentials
[15:15:50] Load layer [main] rpi/misc-skel
[15:15:50] Load layer [main] sys-apps/systemd-net-min
[15:15:50] Load layer [main] sys-apps/fake-hwclock
[15:15:50] /home/pi/rpi-image-gen/device[pre-build.sh]
[15:15:50] /home/pi/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
```

I already created other image config but this armhf version just don't work for me

LorandMarton, 2025-04-23:
@daybyter Yes it works, I just changed the hostname, user, password and image name.

LorandMarton, 2025-05-06:
@daybyter Could you share your working 32-bit configuration? I want to try again this week :)

daybyter, 2025-05-06:
Sorry, but I moved on to a 64 bit config. My customer decided to pay for a pi4 setup.

learmj, 2025-05-10:
Hi. I propose to close this unless there is strong evidence to suggest there is an rpi-image-gen bug that needs investigating.

learmj, 2025-05-10:
It's likely that I'll add a pi3 device directory as part of some upcoming new functionality that I'm working on.

LorandMarton, 2025-05-11:
@learmj From my side, the only remaining question is: what is the correct way to build a 32-bit image?
As you can see in my comments above, I shared a minimal config and followed your suggestions (including checking the UUIDs, image layout, and cmdline.txt), and everything seems to be in order. However, the image still fails to boot.

If there is a known working example for a 32-bit armhf configuration, or any additional hint, I’d really appreciate it.

learmj, 2025-05-12:
I've just successfully built and ran on image on my pi zero 2w using: ```./build.sh -c config/raspbian-bookworm-v7-apt.cfg```.
However, the same image does not boot on my pi4 B. There is a failure very early on at kernel boot ending in:
```
[    0.220968] bcm2835-dma fe007000.dma-controller: Failed to map zero page
[    0.220971] bcm2835-dma fe007000.dma-controller: probe with driver bcm2835-dma failed with error -12
```
I presume you already checked the output of dmesg? Do you see similar on your pi4? What about pi3?

learmj, 2025-05-12:
If you must run a 32-bit kernel, the v7l kernel is needed for pi4 (has LPAE support).
[https://www.raspberrypi.com/documentation/computers/configuration.html#kernel-files-img](url)
The kernel selected by profile/raspbian/bookworm-v7-apt-min is v7 not v7l (there isn't a v7l layer).

I advise you to use a 64-bit kernel (even with 32-bit userland) , which is v8 for both pi3 and pi4.

learmj, 2025-06-09:
I'll likely associate this ticket to device class support (see https://github.com/raspberrypi/rpi-image-gen/issues/64#issuecomment-2915357937).
I anticipate that pi3 will be supported via that scheme.

stu-spp, 2025-07-11:
Sorry to high jack this issue a bit; what about cm5? Is there going to be a specific class for `cm5` or should we continue to use `pi5` when we are building images for the `cm5`?

learmj, 2025-07-14:
Yes. I do plan on supporting additional attributes for device specific images. At the moment, device specific attributes do very little and because (a) there is a generic kernel and firmware installed for most profiles and (b) a generic config.txt installed, your 64-bit image will boot on any device that supports a 64-bit os.

Device specific images have taken a back seat due to (a) LUKS provisioning related work and (b) usability discussions and re-working the rpi-image-gen flow.

learmj, 2025-09-05:
v2.0.0-rc.1 is released. I'm closing this ticket. Please submit new issues as necessary. Thanks.