raspberrypi/rpi-image-gen #42: Feature request: Add a profile for ARMbian

mhalano, 2025-04-06:
My idea is to add a profile for [ARMbian](https://www.armbian.com), another Debian-based system for aarch64 systems. It has support for Raspberry Pi: https://www.armbian.com/rpi4b/.

learmj, 2025-04-07:
While I'm not against supporting third-party distro profiles, rpi-image-gen is designed for users to have their own profiles and have these reside out of tree in a directory of their choosing. The design does not prevent you from doing that today.

ghollingworth, 2025-04-08:
This isn't something we would do...  Also ARMbian already has its own build system so this seems like just a complication...  Not against accepting PRs for such things (assuming they've been implemented in an suitable way), but this isn't something we have much interest in.

closed 2025-04-08 by ghollingworth