raspberrypi/rpi-image-gen #43: Feature request: Add support to F2FS

mhalano, 2025-04-06:
I would like to generate an image using F2FS so my microSD card keeps working for a longer time.

I created a custom config file and set `rootfs_type` under `image` session to `f2fs` and got this:
```
➜ ./build.sh -c config/argon-eon-nas.cfg
[08:36:24] Reading /home/mhalano/code/rpi-image-gen/config/argon-eon-nas.cfg with options []
[08:36:24] IGDEVICE /home/mhalano/code/rpi-image-gen/device/pi4
[08:36:24] IGIMAGE /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual
[08:36:24] IGPROFILE /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[08:36:24] Load profile /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[08:36:24] Load layer [main] debian/bookworm/arm64/base-apt
[08:36:24] Load layer [main] rpi/debian/bookworm/apt
[08:36:24] Load layer [main] rpi/misc-utils
[08:36:24] Load layer [main] rpi/base/essential
[08:36:24] Load layer [main] rpi/boot-firmware
[08:36:24] Load layer [main] rpi/arm64/linux-image-v8
[08:36:24] Load layer [main] rpi/user-credentials
[08:36:24] Load layer [main] rpi/misc-skel
[08:36:24] Load layer [main] sys-apps/systemd-net-min
[08:36:24] Load layer [main] sys-apps/fake-hwclock
[08:36:24] /home/mhalano/code/rpi-image-gen/device[pre-build.sh]
[08:36:24] /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
[08:36:24] Unsupported rootfs type (f2fs).
[08:36:24] Hook Error: [/home/mhalano/code/rpi-image-gen/image/mbr/simple_dual/pre-build.sh] (1)
```

learmj, 2025-04-07:
Hi again. I don't plan on extending the in-tree image profiles to support multiple different types of filesystems. It will end up being a maintenance burden. The image profile you are trying to use supports ext4 and btrfs filesystem types only. If you'd like a custom image profile to support f2fs, it's possible to create a new image layout out of tree and maintain it yourself.

mhalano, 2025-04-08:
You really should consider this, or at least make an option where we can set the filesystem type and it goes all the way to genimage. This filesystem can be very useful in embedded systems running from a microSD card, like Raspberry Pi, or from eMMC, like some CM* modules.
Android itself uses this filesystem for data partition on Android, so it would be nice to know to do serious development.

learmj, 2025-04-08:
> So maybe change this file to just complain about an unsupported file instead of crashing the process?

It doesn't crash. The build is intentionally stopped because the image layout cannot support f2fs (it would need another [third] genimage template).

learmj, 2025-04-08:
@mhalano rpi-image-gen does not prevent you from using other filesystems. All you need to do is copy one of the in-tree image layouts to an external dir, change it as you wish, and invoke build.sh with ```-D``` specifying the path to your image dir.
The same applies to the in-tree profiles, meta layers, device dirs, etc. The tool does not try to be a 'one size fits all'.

closed 2025-04-30 by learmj

mhalano, 2025-04-30:
@learmj sorry, but this was implemented? If not I think you should close the issue as not planned. What do you think?