raspberrypi/rpi-image-gen #44: Add `binfmt-support` as dependency

mhalano, 2025-04-06:
I had to install the package `binfmt-support` because of the command `update-binfmts` that's present in the package `binfmt-support` but it wasn't installed when I run install-deps.sh

learmj, 2025-04-07:
Hi again. As I already wrote in an older ticket (please search tickets before creating new ones), at the current time rpi-image-gen is not stated as supporting running under qemu. I know it works because I've run it successfully. If and when I formally claim that support, I will add the required packages to the deps.

> I think most people would use an amd64 system to generate the image.

IMHO this is an incorrect assumption.

jonnymacs, 2025-04-14:
@learmj fwiw, in testing I'm not seeing binfmt-support is necessary building on an amd machine. On my intel mac, if I do not install binfmt-support and I override the flag in the dependencies_check, I am able to successfully run and build the default image.

The check looks like it dates back many years in the pi-gen project. Perhaps you recall why it was necessary then to support miscellaneous binary formats?

It might be worth considering a warning instead of exit if binfmt_misc is not loaded and arch is not arm. I've currently added some logic to my Dockerfile to override the flag.

Also understand this is not a priority for you, just sharing what I'm learning as I'm helping others out.

Thank you for this awesome utility!

learmj, 2025-04-23:
Hi @jonnymacs Sorry for the late reply. I was curious about this, so I spent a little time looking into it this morning.

> in testing I'm not seeing binfmt-support is necessary building on an amd machine.

[1] [https://docs.docker.com/build/building/multi-platform/](url) is worth a read. Not sure if you're familiar with it.

If I spin up a docker container on my linux x86 box which has [qemu-user-static](https://github.com/multiarch/qemu-user-static) installed, I can see that I'm able to execute many different architectures in the container itself. You can run ```arch-test``` but I recommend running ```docker run --privileged --rm tonistiigi/binfmt``` as it will be more enlightening.

What's happening is that the docker engine is picking up host support for all these architectures and making them available in the container automatically. If I run the same container without ```qemu-user-static``` installed on the x86 box, the result is very different and inline with a system that does not have any 'non-native emulation' installed.

However, it's not all about qemu-user-static... Assuming your host machine is running systemd, you will likely have [systemd-binfmt.service](https://www.freedesktop.org/software/systemd/man/latest/systemd-binfmt.service.html) running and doing it's thing. Even if you have qemu-user-static installed, if you stop this service and try running a non-native arch command in a container, it will not go well, eg
```
$ docker run --rm --platform linux/arm64 -t alpine uname -m
exec /bin/uname: exec format error
```
This is because stopping this service unregisters all ```qemu-*``` formats from under ```/proc/sys/fs/binfmt_misc/```.
Restarting the service shows we have aarch64 support available (bundled with qemu-user-static):
```
$ cat /proc/sys/fs/binfmt_misc/qemu-aarch64 
enabled
interpreter /usr/libexec/qemu-binfmt/aarch64-binfmt-P
flags: POCF
offset 0
magic 7f454c460201010000000000000000000200b700
mask ffffffffffffff00fffffffffffffffffeffffff
```
[https://docs.kernel.org/admin-guide/binfmt-misc.html](url) describes the flags.

I think the following is correct:

1. binfmt_misc tells the kernel how to handle binaries of non-native architectures.
2. qemu static binaries are installed and registered with the system so they can handle execution of non-native binaries transparently.
3. Using docker run with a non-native arch image/program, docker detects the architecture and if qemu is registered, docker silently executes it via qemu. Crucially, this seems like it can also happen _without_  ```--platform``` [1] which surprised me.

So to _run_ non-native programs in a container , clearly we need both binfmt tools (eg deb pkg binfmt-support) and qemu (eg deb pkg qemu-user-static).

To your question about why pi-gen has the check in the dependency... I believe this was all plumbed in to avoid needing a third-party to register the binfmt entry. I think the crucial ticket of interest is https://github.com/RPi-Distro/pi-gen/pull/685.  I suspect it's probably pi-gen's way of decoupling from reliance on the host OS in order to allow it to build an image when running inside a container with arch different to the host.

Hope that helps.

learmj, 2025-04-24:
..and of course if not running in a container, the host definitely does need to have binfmt installed if it won't be able to execute any programs as part of the build (which it will do if building on x86 for an arm/arm64 device). This is obvious, but may have been overlooked by perhaps assuming that non-native execution can only be in a container (which isn't the case).

For example, I can run the default build on my x86 linux box without using Docker, but I absolutely need qemu and binfmt to do so.

I think the check is good to have in (most of the dep checking was carried over from pi-gen) but it can sometimes report incorrectly as you've noticed. Assuming the host does have binfmt installed, the filesystem should probably be mounted as part of container init sequence, eg as part of ```docker run ...```:
```
mount binfmt_misc -t binfmt_misc /proc/sys/fs/binfmt_misc
```
That way the check would pass, both when running the build inside and outside a container.

jonnymacs, 2025-04-27:
Thank you Matthew, this is super helpful.

I'm familiar with multi platform images, but I have not studied the detail in the doc you shared so this has been a good exercise.

I'm still trying to wrap my head around all this. I'm getting there.

[This looks like the important part
](https://docs.docker.com/build/building/multi-platform/#qemu)

> Docker Desktop supports running and building multi-platform images under emulation by default. No configuration is necessary as the builder uses the QEMU that's bundled within the Docker Desktop VM.

Since Im on MacOS using Docker Desktop, I think what's happening is the Linux VM which is running on Apple Virtualization framework, includes qemu-user-static and binfmt_misc, and then as you say "likely this is how docker is picking up host support for all these architectures and making them available in the container automatically"

> If you're using a builder outside of Docker Desktop, such as if you're using Docker Engine on Linux, or a custom remote builder, you need to install QEMU and register the executable types on the host OS. The prerequisites for installing QEMU are:

In your case, running the docker engine directly (no Docker Desktop), you have to manually install qemu-user-static and binfmt-support on the host to get the same effect. If you don't manually install these, or if you uninstall them, then you're toast trying to run anything non native in the container.

So this seems like a bit of Docker Desktop magic (providing qemu and binfmt_misc in the vm host), with additional Docker magic for both of us (docker detects the architecture and if qemu is registered, docker silently executes it via qemu.)

As for the systemd-binfmt.service dependency, again using Docker Desktop, this is abstracted in the Linux VM, so I'm unlikely to stop it.

But again in your case, you need binfmt_misc installed and doing it's thing.

> So to run non-native programs in a container , clearly we need both binfmt tools (eg deb pkg binfmt-support) and qemu (eg deb pkg qemu-user-static).

Yes, but only on the host. If we know we're in a container, running on a host with qemu and binfmt_misc, we don't need to install qemu or mount binfmt_misc in the container. Hence the check is unnecessary in my case.

> and of course if not running in a container, the host definitely does need to have binfmt installed 

right. host needs it in both cases - seems like the common ground.

>I think the check is good to have in

I tend to agree after these learnings, however I do wish it could know things are ok for the case where the host has binfmt_misc support and /proc/sys/fs/binfmt_misc is _not_ mounted. Experimenting a little... what about would you consider this check instead?

```
cat /proc/filesystems | grep binfmt_misc
```

>Assuming the host does have binfmt installed, the filesystem should probably be mounted as part of container init sequence
>mount binfmt_misc -t binfmt_misc /proc/sys/fs/binfmt_misc

I'm not sure I agree since it doesn't need to be mounted if it's installed on the host. The command requires sudo and adds complexity to the start up.

---------------

Last thing - for posterity since it was helpful for me to understand all this - once I have a container, I run rpi-image-gen which runs podman and my container becomes the host for the podman instance. I still do not need binfmt-support in the container, but I do need to install qemu-user-static or podman fails.

The explanation - is that binfmt-support is a Linux Kernel feature and therefore it's inherited as I create containers in containers. But qemu is not, and I need to have the compiled binary on each of the Russian doll hosts.

learmj, 2025-04-28:
Hi @jonnymacs 

> Yes, but only on the host.  If we know we're in a container, running on a host with qemu and binfmt_misc, we don't need to install qemu or mount binfmt_misc in the container. Hence the check is [...]

Agreed, although I don't know how much of this 'auto magic' is Docker specific.

> I tend to agree after these learnings, however I do wish it could know things are ok for the case where the host has binfmt_misc support and /proc/sys/fs/binfmt_misc is _not_ mounted. Experimenting [...]

I think there needs to be a prior check to ascertain if running in a container (or not) before that. That sort of check has evolved over the years. Solutions that used to work now seem not to, and those that do work seem a little bit particular to the engine involved.
 
> I'm not sure I agree since it doesn't need to be mounted if it's installed on the host. The command requires sudo and adds complexity to the start up.

Yes, but only if the build is running in a container. And I don't know how much of that is Docker specific.

Also, you probably need flags like ```--privileged``` and ```--cap-add=ALL``` in ```docker run``` (I know I do on linux) in order to successfully get the rpi-image-gen ```unshare``` working via podman. I don't _think_ those flags can be set without sudo / elevated privs. Do you not require those when launching your docker container on macos?

Hopefully by now it's clear why I've deliberately stayed clear from officially declaring support of running rpi-image-gen non-natively or in a container. It's a bit of a rabbit hole. The pi-gen view on building in a container is that it's community maintained thing. If I add support for running rpi-image-gen in a container it should either be as generic as possible (which will no doubt bring lots of complexity) or be particular to container engine and/or host OS. The former would likely require significant effort and the latter won't suit everyone. There are other things that rpi-image-gen needs to support (a lot of which are on our RPi roadmap) so the focus is on those for the time being.

jonnymacs, 2025-04-29:
>I think there needs to be a prior check to ascertain if running in a container (or not) before that.

I see. I'm reading this as "when _not_ in a container, binfmt_misc _does have to be_ mounted at /proc/sys/fs/binfmt_misc". I suppose this makes sense from [this doc ](https://docs.kernel.org/admin-guide/binfmt-misc.html)

In that case I feel better about doing the overrides in the Dockerfile as we know we'll be in a container when the image runs.

> Hopefully by now it's clear why I've deliberately stayed clear from officially declaring support of running rpi-image-gen non-natively or in a container. It's a bit of a rabbit hole.

Ha! 😄 🐰 👍 A rabbit hole indeed. That was never in question from my pov. I understand this completely. Keeping your focus on building a tool that is fully tested and guaranteed to work on the actual rpi hw with the official rpi os seems like the right call to me. If the rest of us want to _try_ to make it work elsewhere - we are welcome, but no guarantees.

I do appreciate the time and wisdom given your experience and expertise though, and may come back with more questions as time goes on. So please don't feel pressure to respond right away if I do.

closed 2025-09-05 by learmj