raspberrypi/rpi-image-gen #45: Problem with GPG key during the build

mhalano, 2025-04-06:
I executed a simple build and got this error. I couldn't find how to solve it. I even installed `debian-keyring` package manually (I'm using Ubuntu).
```
➜ ./build.sh
[08:56:56] Reading /home/mhalano/code/rpi-image-gen/config/generic64-apt-simple.cfg with options []
[08:56:57] IGDEVICE /home/mhalano/code/rpi-image-gen/device/pi5
[08:56:57] IGIMAGE /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual
[08:56:57] IGPROFILE /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[08:56:57] Load profile /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[08:56:57] Load layer [main] debian/bookworm/arm64/base-apt
[08:56:57] Load layer [main] rpi/debian/bookworm/apt
[08:56:57] Load layer [main] rpi/misc-utils
[08:56:57] Load layer [main] rpi/base/essential
[08:56:57] Load layer [main] rpi/boot-firmware
[08:56:57] Load layer [main] rpi/arm64/linux-image-v8
[08:56:57] Load layer [main] rpi/user-credentials
[08:56:57] Load layer [main] rpi/misc-skel
[08:56:57] Load layer [main] sys-apps/systemd-net-min
[08:56:57] Load layer [main] sys-apps/fake-hwclock
[08:56:57] /home/mhalano/code/rpi-image-gen/device[pre-build.sh]
[08:56:57] /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
I: automatically chosen mode: root
I: arm64 cannot be executed natively, but transparently using qemu-user binfmt emulation
I: automatically chosen format: directory
I: running --setup-hook in shell: sh -c 'bin/runner setup "$@"' exec /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/rootfs
I: running apt-get update...
done
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Err:1 http://deb.debian.org/debian bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131 NO_PUBKEY F8D2585B8783D481
Err:2 http://deb.debian.org/debian-security bookworm-security InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 54404762BBB6E853 NO_PUBKEY BDE6D2B9216EC7A8
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Err:3 http://deb.debian.org/debian bookworm-updates InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131
Get:4 http://archive.raspberrypi.com/debian bookworm InRelease [39.3 kB]
Err:4 http://archive.raspberrypi.com/debian bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists...
W: http://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: OpenPGP signature verification failed: http://deb.debian.org/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131 NO_PUBKEY F8D2585B8783D481
E: The repository 'http://deb.debian.org/debian bookworm InRelease' is not signed.
W: http://deb.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: OpenPGP signature verification failed: http://deb.debian.org/debian-security bookworm-security InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 54404762BBB6E853 NO_PUBKEY BDE6D2B9216EC7A8
E: The repository 'http://deb.debian.org/debian-security bookworm-security InRelease' is not signed.
W: http://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: OpenPGP signature verification failed: http://deb.debian.org/debian bookworm-updates InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0E98404D386FA1D9 NO_PUBKEY 6ED0E7B82643E131
E: The repository 'http://deb.debian.org/debian bookworm-updates InRelease' is not signed.
W: http://archive.raspberrypi.com/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: OpenPGP signature verification failed: http://archive.raspberrypi.com/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.com/debian bookworm InRelease' is not signed.
E: apt-get update --error-on=any -oAPT::Status-Fd=<$fd> -oDpkg::Use-Pty=false failed
W: hooklistener errored out: E: received eof on socket

I: main() received signal PIPE: waiting for setup...
E: mmdebstrap failed to run
2025-04-06 08:56:58,427 bdebstrap ERROR: mmdebstrap failed with exit code 29. See above for details.
[08:56:58] Error: [podman unshare bdebstrap --config /home/mhalano/code/rpi-image-gen/meta/debian/bookworm/arm64/base-apt.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/debian/bookworm/apt.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/misc-utils.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/base/essential.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/boot-firmware.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/arm64/linux-image-v8.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/user-credentials.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/misc-skel.yaml --config /home/mhalano/code/rpi-image-gen/meta/sys-apps/systemd-net-min.yaml --config /home/mhalano/code/rpi-image-gen/meta/sys-apps/fake-hwclock.yaml --env IGconf_device_class=pi5 --env IGconf_device_hostname=raspberrypi --env IGconf_device_keyboard_keymap=gb --env IGconf_device_keyboard_layout=English (UK) --env IGconf_device_locale1=en_GB.UTF-8 --env IGconf_device_profile=bookworm/apt-min64 --env IGconf_device_ssh_user1=n --env IGconf_device_storage_type=sd --env IGconf_device_timezone=Europe/London --env IGconf_device_timezone_area=Europe --env IGconf_device_timezone_city=London --env IGconf_device_user1=pi --env IGconf_device_variant=none --env IGconf_image_boot_part_size=200% --env IGconf_image_compression=none --env IGconf_image_layout=mbr/simple_dual --env IGconf_image_name=deb12-arm64-min --env IGconf_image_root_part_size=300% --env IGconf_image_rootfs_type=ext4 --env IGconf_image_suffix=img --env IGconf_image_version=2025-04-06 --env IGconf_sbom_output_format=spdx-json --aptopt APT::Get::Purge true --aptopt Dir::Etc::TrustedParts /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_apt_keydir=/home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_deploydir=/home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/deploy --env IGconf_sys_outputdir=/home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/artefacts --env IGconf_sys_target=/home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/rootfs --env IGconf_sys_workdir=/home/mhalano/code/rpi-image-gen/work/deb12-arm64-min --env IGTOP=/home/mhalano/code/rpi-image-gen --env META_HOOKS=/home/mhalano/code/rpi-image-gen/meta-hooks --env RPI_TEMPLATES=/home/mhalano/code/rpi-image-gen/templates/rpi --env IGDEVICE=/home/mhalano/code/rpi-image-gen/device/pi5 --env IGIMAGE=/home/mhalano/code/rpi-image-gen/image/mbr/simple_dual --env IGPROFILE=/home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64 --env PATH=/home/mhalano/code/rpi-image-gen/bin:/home/mhalano/.rbenv/shims:/home/mhalano/.rbenv/bin:/home/mhalano/.rd/bin:/home/mhalano/.local/bin:/home/mhalano/bin:/home/mhalano/.cargo/bin:/home/mhalano/.npm-global/bin:/home/mhalano/.krew/bin:/home/mhalano/.tfenv/bin:/home/mhalano/.local/share/antidote/https-COLON--SLASH--SLASH-github.com-SLASH-romkatv-SLASH-zsh-bench:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin --force --name deb12-arm64-min --hostname raspberrypi --output /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/artefacts --target /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/rootfs --setup-hook bin/runner setup "$@" --essential-hook bin/runner essential "$@" --customize-hook bin/runner customize "$@" --cleanup-hook bin/runner cleanup "$@"] (1)
```

mhalano, 2025-04-06:
In the end, I installed the package `debian-archive-keyring`, not just `debian-keyring`, and did work, so maybe it should be part of install-deps.sh.

mhalano, 2025-04-06:
I found other error, though:
```
➜ ./build.sh
[09:12:45] Reading /home/mhalano/code/rpi-image-gen/config/generic64-apt-simple.cfg with options []
[09:12:45] IGDEVICE /home/mhalano/code/rpi-image-gen/device/pi5
[09:12:45] IGIMAGE /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual
[09:12:45] IGPROFILE /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[09:12:46] Load profile /home/mhalano/code/rpi-image-gen/profile/bookworm/apt-min64
[09:12:46] Load layer [main] debian/bookworm/arm64/base-apt
[09:12:46] Load layer [main] rpi/debian/bookworm/apt
[09:12:46] Load layer [main] rpi/misc-utils
[09:12:46] Load layer [main] rpi/base/essential
[09:12:46] Load layer [main] rpi/boot-firmware
[09:12:46] Load layer [main] rpi/arm64/linux-image-v8
[09:12:46] Load layer [main] rpi/user-credentials
[09:12:46] Load layer [main] rpi/misc-skel
[09:12:46] Load layer [main] sys-apps/systemd-net-min
[09:12:46] Load layer [main] sys-apps/fake-hwclock
[09:12:46] /home/mhalano/code/rpi-image-gen/device[pre-build.sh]
[09:12:46] /home/mhalano/code/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
I: automatically chosen mode: root
I: arm64 cannot be executed natively, but transparently using qemu-user binfmt emulation
I: finding correct signed-by value...
done
I: automatically chosen format: directory
I: running --setup-hook in shell: sh -c 'bin/runner setup "$@"' exec /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/rootfs
I: running apt-get update...
done
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 Packages [8692 kB]
Get:5 http://archive.raspberrypi.com/debian bookworm InRelease [39.3 kB]
Get:6 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5832 B]
Get:7 http://deb.debian.org/debian bookworm/non-free arm64 Packages [75.8 kB]
Get:8 http://deb.debian.org/debian bookworm/contrib arm64 Packages [45.6 kB]
Get:9 http://deb.debian.org/debian-security bookworm-security/contrib arm64 Packages [508 B]
Get:10 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [247 kB]
Get:11 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [512 B]
Err:5 http://archive.raspberrypi.com/debian bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists...
W: http://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/raspberrypi-archive-stable.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/raspberrypi-archive-stable.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/raspberrypi-archive-stable.gpg are ignored as the file has an unsupported filetype.
W: http://archive.raspberrypi.com/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/google-cloud-cli-archive-keyring.gpg are ignored as the file has an unsupported filetype.
W: http://archive.raspberrypi.com/debian/dists/bookworm/InRelease: The key(s) in the keyring /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/keys/raspberrypi-archive-stable.gpg are ignored as the file has an unsupported filetype.
W: OpenPGP signature verification failed: http://archive.raspberrypi.com/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.com/debian bookworm InRelease' is not signed.
E: apt-get update --error-on=any -oAPT::Status-Fd=<$fd> -oDpkg::Use-Pty=false failed
W: hooklistener errored out: E: received eof on socket

I: main() received signal PIPE: waiting for setup...
E: mmdebstrap failed to run
2025-04-06 09:12:47,978 bdebstrap ERROR: mmdebstrap failed with exit code 29. See above for details.
[09:12:47] Error: [podman unshare bdebstrap --config /home/mhalano/code/rpi-image-gen/meta/debian/bookworm/arm64/base-apt.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/debian/bookworm/apt.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/misc-utils.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/base/essential.yaml --config /home/mhalano/code/rpi-image-gen/meta/rpi/boot-firmware.yaml --co [... cut ...] r/local/games:/snap/bin --force --name deb12-arm64-min --hostname raspberrypi --output /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/artefacts --target /home/mhalano/code/rpi-image-gen/work/deb12-arm64-min/rootfs --setup-hook bin/runner setup "$@" --essential-hook bin/runner essential "$@" --customize-hook bin/runner customize "$@" --cleanup-hook bin/runner cleanup "$@"] (1)
☸ minikube (default) in rpi-image-gen on  master [?] on ☁️  (us-east-1) on ☁️  marcoshalano@gmail.com took 2s
```

mhalano, 2025-04-06:
The same problem happens with pi-gen: https://github.com/RPi-Distro/pi-gen/issues/668

learmj, 2025-04-07:
Hi again. Please search tickets before creating new ones. 
https://github.com/raspberrypi/rpi-image-gen/issues/24

You can specify the keys you'd like rpi-image-gen to use.

learmj, 2025-04-08:
The key does not need to be dearmored if using up-to-date APT (although probably old now - see link below).
Looks like a lot (all?) the keys you are using are complained about, not just the rpi key. Implies more of a system wide problem IMHO.

[https://wiki.debian.org/DebianRepository/UseThirdParty#OpenPGP_Key_distribution](url)

> The reason why we avoid ASCII-Armored files is that they can only be used by [SecureApt](https://wiki.debian.org/SecureApt) in version 1.4 or later (which appeared in stretch). We also strongly recommend the use of HTTPS as it bypasses certain MITM attacks that would allow a hostile third party to inject OpenPGP certificate material in the repository setup.

There's nothing that needs fixing here.

mhalano, 2025-04-08:
Strange, because I'm doing a simple `./build.sh -c config/generic64-apt-simple.cfg`, so theoretically it should work out of the box.

learmj, 2025-04-08:
Indeed, but you're running under qemu which as I've previously mentioned, does not have claimed official support.
> I: arm64 cannot be executed natively, but transparently using qemu-user binfmt emulation

mhalano, 2025-04-08:
It doesn't seem to be related, since with QEMU and dearmored key the process works.
There are any restrictions about the OS I must use, even on ARM? Like Ubuntu wouldn't have this GPG key added natively, just Raspi OS.

learmj, 2025-04-08:
Correct. The user has control over the keys they use with the host apt.
As per the [readme](https://github.com/raspberrypi/rpi-image-gen/#overview):

> rpi-image-gen heavily favours a Debian-ish installation process where a temporary apt configuration is assembled and used to seed the chroot. 

This implies that it requires a host machine where apt is available (it will be invoked by ```mmdebstrap```).

Also: see the description of the [IGconf_sys_apt_keydir](https://github.com/raspberrypi/rpi-image-gen/blob/master/sys-build.defaults#L26)