raspberrypi/rpi-image-gen #46: Download with apt problem : permission denied to mount proc and sys directory

daybyter, 2025-04-16:
Hi!

I customized your webkiosk example and changed the customize script to download some package. Once they are exracting, I get errors, that they cannot mount the work/<img_name>/rootfs/proc and sys directory. Is this a misconfiguration from me?

learmj, 2025-04-23:
Hi. Thanks for your report. If you're building in a container then this is likely a problem with your container environment / configuration (I have seen similar if I spin up a simple Docker container on my x86 linux box, git clone and run the default build). If not then messages like the following are normal:

```
mount: /path/to/work/deb12-arm64-min/rootfs/proc: permission denied.
       dmesg(1) may have more information after failed mount system call.
W: since mounting /proc normally failed, /proc is now bind-mounted instead

```
These messages are from ```mmdebstrap``` as it prepares the chroot. It falls back to bind mounting /proc if it's unable to mount it directly. They are normal messages and nothing to be concerned about.

> Once they are exracting, I get errors

The default build runs fine on an RPiOS host (I typically use a Pi5 with NVMe drive).
As I've mentioned elsewhere, building inside a container env or via emulation is not formally supported. Others have had success building under emulation.

learmj, 2025-05-10:
Hi. I intend on closing this because as far as I can tell, the problem isn't caused by rpi-image-gen.

stu-spp, 2025-05-10:
Agreed. I build in a (Debian aarch64) container, and see these warnings, and the builds succeed without issue. I think it's safe to close this.

closed 2025-05-12 by learmj