raspberrypi/rpi-image-gen #47: Build image with wifi support

benjibst, 2025-04-23:
In the default configuration files, only wired networking is supported. How to build an image for the Zero 2 W with wifi support, and how can the default network be configured, like in raspberry pi imager?

learmj, 2025-04-23:
Hi. Thanks for your report. You're absolutely right. At the moment, the default build is very simple and has wired networking only. Furthermore, there is actually no wlan or BT firmware installed anywhere for any build target/device. No meta layer exists for it and no package (eg ```firmware-brcm80211```) is even installed.

I intend on moving linux + firmware packages from the top level profile down to a device level profile so that assets installed by, for example ```raspi-firmware```, can be fine tuned for the device the build is targeting. I intend this to be similar for ```firmware-brcm80211``` (ie, if your build is targeting a Pi5 you only need boot firmware files and wifi/BT assets relevant to a Pi5). Once this happens, packages installed which are specific to a device (think of it like a BSP) will be decoupled from the top level profile.

Do bear in mind that if you know you targeting a zero2w, you can have your own out-of-tree directory, meta layers and profile to do exactly what you need (ie install ```firmware-brcm80211``` and your wlan supplicant of choice, plus start-up scripts, etc).

Does that help?

benjibst, 2025-04-23:
At the moment I switched to the RPI4B because it has an ethernet port and this will simplify things. After building an image, i flashed the rootfs to a second partition and changed cmdline.txt in order to boot to that partition. Inspecting the serial console output during boot makes me believe the kernel is booting successfully, but i can not access the PI through SSH. is this because the ssh package is not installed on the image, or because the network is not working, or something different? im using the example from the repository.

benjibst, 2025-04-23:
Thanks to everyone for the help, i was able to boot into the new partition and log in via SSH.

closed 2025-04-23 by learmj