raspberrypi/rpi-image-gen #50: Invalid image sizes when using dynamic partition size

martin-ottens, 2025-04-30:
I have built an image with the `mbr/simple_dual` config, ext4 root file system, and set `root_part_size` to 120%.
The output image is completely OK, but there are cases where the size of the image is not a multiple of 512 bytes, so that the Raspberry Pi Imager, for example, reports an error.

A short example:
```
$ wc -c < output.img
1504609894

$ fdisk -l output.img
Disk output.img: 1.4 GiB, 1504609792 bytes, 2938691 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device      Boot  Start     End Sectors  Size Id Type
output.img1 *     16384  114687   98304   48M  c W95 FAT32 (LBA)
output.img2      114688 2949119 2834432  1.4G 83 Linux
```
The size of the image file is `1504609894` bytes, but only `1504609792` bytes are used, 102 bytes are unused and have zero values (also `1504609894 % 512 = 102`).
I suspect that this is actually a bug in genimage, but I'm surprised that the problem doesn't seem to have appeared there yet.

After the image creation is completed, I've implemented the following workaround in my workflow:
```
truncate -s %512 <IMAGE.IMG>
```
This adds zeros to the end of the image file so that a multiple of 512 bytes is reached. This could also be integrated into the post-image hooks if there is no other/better solution.

learmj, 2025-04-30:
Hi @martin-ottens Thanks very much for the report. I occasionally use rpi-imager to write an image, but these days I typically use [pi-gen-micro's](https://github.com/raspberrypi/pi-gen-micro) fastboot gadget and use a fastboot client to write the sparse image that rpi-image-gen creates to the device. I've not seen this problem before when I've Imager. The genimage ```%``` sizing is a useful feature, but it's very approximate and I can well believe it could lead to what you describe...

> Disk output.img: 1.4 GiB, 1504609792 bytes, 2938691 sectors
..
> The size of the image file is 1504609894 bytes, but only 1504609792 bytes are used
 
...which sounds like the image file generated by genimage _is_ a multiple of 512 but it's the partition table that isn't making use of all the space (because it's 410 bytes short, most likely as a result of the percentage sizing algo). That's why fdisk reports 1504609792 bytes. Maybe the genimage sizing algo in the version you're using isn't (doesn't?) round up to the intended block size of the image (ie 512 in your case). I'd have to look. What version of genimage are you using?

> also 1504609894 % 512 = 102.

Normal. genimage will pad the hdimage up to the block size boundary.

> I suspect that this is actually a bug in genimage, but I'm surprised that the problem doesn't seem to have appeared there yet.

Not sure I agree. genimage appears to be doing the right thing. If you use ```fill``` in your ```hdimage``` section, does the behaviour change? You'll have to specify a size there too. Try specifying a size of 1504609894 to keep things consistent then increase it by 410 and see if things align.

> This could also be integrated into the post-image hooks if there is no other/better solution.

Maybe, but it _feels_ like the problem is elsewhere. Other attributes will change as a result of using gpt so it's not a like for like test changing your ```partition-table-type = "mbr"``` to gpt. Performing any truncation post-image could risk wiping out the backup GPT (although I'd expect that to end on a sector boundary, so maybe it would be safe to 'trim' the residual bytes).

martin-ottens, 2025-05-02:
Hi,
thanks for the fast reply.

> What version of genimage are you using?

Im running the latest version of Raspberry Pi OS, apt installs genimage version 16. I have also tested with version 18, which is the latest version - the issue persists.

> Not sure I agree. genimage appears to be doing the right thing. If you use fill in your hdimage section, does the behaviour change?

When changing the `hdimage` section to
```
hdimage {
    align = 8M
    partition-table-type = "mbr"
    fill = true
}
```
the image has the expected size with `% 512 == 0`. But, as written in the genimage [docs](https://github.com/pengutronix/genimage/blob/v18/README.rst?plain=1#L424), it increases the size of the image, im my case:
| Image type | without `fill`  | with `fill = true` | difference (bytes) |
|------------------|--------------------|-----------------------|-----------------|
| Normal        | 1505710899 | 1509949440      | +4238541  |
| Sparse         | 1183116240 | 1183116240      | +0                |

The difference is less than the 25% selected additional partition size.
I think adding `fill = true` to the examples is the best solution. Since this is only relevant for percentage size specifications, you could also activate this dynamically  in `pre-image.sh` so that it is always ensured that the images also work with rpi-imager. I am happy to provide a PR if you like.

> Performing any truncation post-image could risk wiping out the backup GPT (although I'd expect that to end on a sector boundary, so maybe it would be safe to 'trim' the residual bytes).

Using `truncate -s %512 <IMAGE.IMG>` is still a valid workaround. From the man page:
> SIZE may also be prefixed by one of the following modifying characters: [...] '%' round up to multiple of.

So there are no bytes trimmed, just a maximum of 511 zero bytes are appended to the file.

learmj, 2025-05-02:
Hi. Thanks for the update and the points made. I agree with you.

Re: fill

> it increases the size of the image

I think it would be less if ```align``` was smaller though, and would probably still divide equally by 512. I would expect that ```fill=true``` takes ```align``` into account. I'd also expect no increase to the sparse size as a result of using fill (as the image is essentially just padded) which would be invisible in the sparse variant.

> Since this is only relevant for percentage size specifications

There's much more chance in the image not being divisible by 512 if using percentages (without fill), but whether it happens or not isn't strictly limited to using percentage sizes.

> Using truncate -s %512 <IMAGE.IMG> is still a valid workaround.

I think adding a simple ```truncate -s %512 /path/to/output.img``` in the post-image hook is the way to go.
If we can make this simple change to keep Imager sweet, I'm fine with it.

The truncate invocation described won't modify the image if its size is already a multiple of 512 and as genimage already make sure that partition alignment obeys ```align``, this is probably a good 'interoperability' change to make for primitive tools that expect the input file to be a multiple of the sector size. I wonder if we should bump it to 4096.

I can make the change to post-image - no need for a PR although the offer is appreciated.

closed 2025-05-07 by martin-ottens