raspberrypi/rpi-image-gen #52: Possible issues with the keydir/ directory contents

ardavast, 2025-05-05:
Building on latest Debian 12 failed:
```
[10:26:42] Reading /home/admin/rpi-image-gen/config/generic64-apt-simple.cfg with options []
[10:26:42] IGDEVICE /home/admin/rpi-image-gen/device/pi5
[10:26:42] IGIMAGE /home/admin/rpi-image-gen/image/mbr/simple_dual
[10:26:42] IGPROFILE /home/admin/rpi-image-gen/profile/bookworm/apt-min64
[10:26:42] Load profile /home/admin/rpi-image-gen/profile/bookworm/apt-min64
[10:26:42] Load layer [main] debian/bookworm/arm64/base-apt
[10:26:42] Load layer [main] rpi/debian/bookworm/apt
[10:26:42] Load layer [main] rpi/misc-utils
[10:26:42] Load layer [main] rpi/base/essential
[10:26:42] Load layer [main] rpi/boot-firmware
[10:26:42] Load layer [main] rpi/arm64/linux-image-v8
[10:26:42] Load layer [main] rpi/user-credentials
[10:26:42] Load layer [main] rpi/misc-skel
[10:26:42] Load layer [main] sys-apps/systemd-net-min
[10:26:42] Load layer [main] sys-apps/fake-hwclock
[10:26:42] /home/admin/rpi-image-gen/device[pre-build.sh]
[10:26:42] /home/admin/rpi-image-gen/image/mbr/simple_dual[pre-build.sh]
WARN[0000] The cgroupv2 manager is set to systemd but there is no systemd user session available 
WARN[0000] For using systemd, you may need to login using an user session 
WARN[0000] Alternatively, you can enable lingering with: `loginctl enable-linger 1000` (possibly as root) 
WARN[0000] Falling back to --cgroup-manager=cgroupfs    
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_ADDRESS = "bg_BG.UTF-8",
	LC_NAME = "bg_BG.UTF-8",
	LC_MONETARY = "bg_BG.UTF-8",
	LC_PAPER = "bg_BG.UTF-8",
	LC_IDENTIFICATION = "bg_BG.UTF-8",
	LC_TELEPHONE = "bg_BG.UTF-8",
	LC_MEASUREMENT = "bg_BG.UTF-8",
	LC_TIME = "bg_BG.UTF-8",
	LC_NUMERIC = "bg_BG.UTF-8",
	LANG = "C.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("C.UTF-8").
I: automatically chosen mode: root
I: chroot architecture arm64 is equal to the host's architecture
I: finding correct signed-by value...
done
I: automatically chosen format: directory
I: running --setup-hook in shell: sh -c 'bin/runner setup "$@"' exec /home/admin/rpi-image-gen/work/deb12-arm64-min/rootfs
I: running apt-get update...
done
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5832 B]
Get:5 http://deb.debian.org/debian bookworm/main arm64 Packages [8692 kB]
Get:6 http://archive.raspberrypi.com/debian bookworm InRelease [39.2 kB]
Get:7 http://deb.debian.org/debian bookworm/non-free arm64 Packages [75.8 kB]
Get:8 http://deb.debian.org/debian bookworm/contrib arm64 Packages [45.6 kB]
Get:9 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [254 kB]
Get:10 http://deb.debian.org/debian-security bookworm-security/contrib arm64 Packages [508 B]
Get:11 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [512 B]
Err:6 http://archive.raspberrypi.com/debian bookworm InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
Reading package lists...
W: GPG error: http://archive.raspberrypi.com/debian bookworm InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 82B129927FA3303E
E: The repository 'http://archive.raspberrypi.com/debian bookworm InRelease' is not signed.
E: apt-get update --error-on=any -oAPT::Status-Fd=<$fd> -oDpkg::Use-Pty=false failed
I: main() received signal PIPE: waiting for setup...
E: mmdebstrap failed to run
2025-05-05 10:26:44,719 bdebstrap ERROR: mmdebstrap failed with exit code 29. See above for details.
[10:26:44] Error: [podman unshare bdebstrap --config /home/admin/rpi-image-gen/meta/debian/bookworm/arm64/base-apt.yaml --config /home/admin/rpi-image-gen/meta/rpi/debian/bookworm/apt.yaml --config /home/admin/rpi-image-gen/meta/rpi/misc-utils.yaml --config /home/admin/rpi-image-gen/meta/rpi/base/essential.yaml --config /home/admin/rpi-image-gen/meta/rpi/boot-firmware.yaml --config /home/admin/rpi-image-gen/meta/rpi/arm64/linux-image-v8.yaml --config /home/admin/rpi-image-gen/meta/rpi/user-credentials.yaml --config /home/admin/rpi-image-gen/meta/rpi/misc-skel.yaml --config /home/admin/rpi-image-gen/meta/sys-apps/systemd-net-min.yaml --config /home/admin/rpi-image-gen/meta/sys-apps/fake-hwclock.yaml --env IGconf_device_class=pi5 --env IGconf_device_hostname=raspberrypi --env IGconf_device_keyboard_keymap=gb --env IGconf_device_keyboard_layout=English (UK) --env IGconf_device_locale1=en_GB.UTF-8 --env IGconf_device_profile=bookworm/apt-min64 --env IGconf_device_ssh_user1=n --env IGconf_device_storage_type=sd --env IGconf_device_timezone=Europe/London --env IGconf_device_timezone_area=Europe --env IGconf_device_timezone_city=London --env IGconf_device_user1=pi --env IGconf_device_variant=none --env IGconf_image_boot_part_size=200% --env IGconf_image_compression=none --env IGconf_image_layout=mbr/simple_dual --env IGconf_image_name=deb12-arm64-min --env IGconf_image_root_part_size=300% --env IGconf_image_rootfs_type=ext4 --env IGconf_image_suffix=img --env IGconf_image_version=2025-05-05 --env IGconf_sbom_syft_config=syft.yaml --aptopt APT::Get::Purge true --aptopt Dir::Etc::TrustedParts /home/admin/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_apt_keydir=/home/admin/rpi-image-gen/work/deb12-arm64-min/keys --env IGconf_sys_deploydir=/home/admin/rpi-image-gen/work/deb12-arm64-min/deploy --env IGconf_sys_outputdir=/home/admin/rpi-image-gen/work/deb12-arm64-min/artefacts --env IGconf_sys_target=/home/admin/rpi-image-gen/work/deb12-arm64-min/rootfs --env IGconf_sys_workdir=/home/admin/rpi-image-gen/work/deb12-arm64-min --env IGTOP=/home/admin/rpi-image-gen --env META_HOOKS=/home/admin/rpi-image-gen/meta-hooks --env RPI_TEMPLATES=/home/admin/rpi-image-gen/templates/rpi --env IGDEVICE=/home/admin/rpi-image-gen/device/pi5 --env IGIMAGE=/home/admin/rpi-image-gen/image/mbr/simple_dual --env IGPROFILE=/home/admin/rpi-image-gen/profile/bookworm/apt-min64 --env PATH=/home/admin/rpi-image-gen/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games --force --name deb12-arm64-min --hostname raspberrypi --output /home/admin/rpi-image-gen/work/deb12-arm64-min/artefacts --target /home/admin/rpi-image-gen/work/deb12-arm64-min/rootfs --setup-hook bin/runner setup "$@" --essential-hook bin/runner essential "$@" --customize-hook bin/runner customize "$@" --cleanup-hook bin/runner cleanup "$@"] (1)
```

I was able to fix this by copying the file `/usr/share/keyrings/raspberrypi-archive-stable.gpg` from a real rpi to the keydir/ directory.
It seems that the same key exists inside the keydir directory with the name `raspberrypi-archive-stable.gpg.key` but it is in an older format, maybe that's why it doesn't work? The other file inside the keydir/ directory, `raspbian-archive-keyring.gpg` also doesn't match, on my rpi the files `raspbian-archive-keyring.gpg` and `raspberrypi-archive-stable.gpg` are the exact same file.

stu-spp, 2025-05-05:
I ran into this problem yesterday on a clean Debian install. I fixed it by:

```shell
gpg --keyserver keyserver.ubuntu.com --recv-keys CF8A1AF502A2AA2D763BAE7E82B129927FA3303E
gpg --export CF8A1AF502A2AA2D763BAE7E82B129927FA3303E | tee /usr/share/keyrings/raspberrypi-archive-keyring.gpg
```

learmj, 2025-05-06:
Hi @ardavast @stu-spp 
rpi-image-gen will happily bootstrap on Raspberry Pi OS because the RPi repo key is installed as non-ascii armored file ```/usr/share/keyrings/raspberrypi-archive-stable.gpg```. I've not really spent much time ensuring the build will run on other distros/platform/OSs, although of course it would be advantageous to support as many as possible.

 On a non RPiOS system, ```/usr/share/keyrings/raspberrypi-archive-stable.gpg``` won't exist - that's why I added the required key in ```keydir/```. However, the problem is that the key required (which is ascii-armored in the repo as it stands) is not accepted by APT because it doesn't end with ```.asc```. This is a mistake on my part which has gone unnoticed (because I always build on Raspberry Pi OS and rpi-image-gen copies the contents of /usr/share/keyrings/ into the working keydir that apt is told to use via ```--aptopt Dir::Etc::TrustedParts``` when creating the chroot).

apt-key:

> Make sure to use the "asc" extension for ASCII armored keys and the "gpg" extension for the binary OpenPGP format (also known as "GPG key public ring"). The binary OpenPGP format works for all apt versions, while the ASCII armored format works for apt version >= 1.4.

If I...

Rename ```keydir/raspberrypi-archive-stable.gpg.key``` -> ```keydir/raspberrypi-archive-stable.asc```
or
De-armor ```keydir/raspberrypi-archive-stable.gpg.key``` -> ```keydir/raspberrypi-archive-stable.gpg```

...I can bootstrap fine without any raspberrypi keys in ```/usr/share/keyrings/```.

One of these is the change I need to make to resolve this problem. The latter is likely better because it's supported for all apt versions (even though I hope a user is building on an up-to-date OS). De-armoring results in the equivalent key @stu-spp mentions above.

learmj, 2025-05-06:
If you verify that https://github.com/raspberrypi/rpi-image-gen/pull/53 fixes this, I'll merge it.

stu-spp, 2025-05-06:
@learmj Yes, your issue/52 branch fixes it! Thank you!!!

learmj, 2025-05-06:
Excellent :-) Thanks!

PR #53 (2025-05-06): Fix RPi apt key installation
On non Raspberry Pi OS build hosts, unless the RPi key is provided to apt, it will fail to authenticate the repo. rpi-image-gen does cater for this. However, the key is not accepted by apt because it's currently stored as ascii-armored but does not have a .asc suffix.

Solutions are to rename it to have a .asc suffix, or de-armor and rename it so it has a .gpg suffix.
The latter is preferred for wider interoperability reasons as per apt-key(8). Doing so results in the same key that would be obtained from (example):

gpg --keyserver keyserver.ubuntu.com --recv-keys  [... cut]
commit aafaaa7a (2025-05-06): Fix RPi apt key installation

On non Raspberry Pi OS build hosts, unless the RPi key is provided to
apt, it will fail to authenticate the repo. rpi-image-gen does cater
for this. However, the key is not accepted by apt because it's
currently stored as ascii-armored but does not have a .asc suffix.

Solutions are to rename it to have a .asc suffix, or de-armor and
rename it so it has a .gpg suffix.
The latter is preferred for wider interoperability reasons as per
apt-key(8). Doing so results in the same key that would be obtained
from (example):

gpg --keyserver keyserver.ubuntu.com --recv-keys [... cut]
closed 2025-05-06 by learmj