raspberrypi/rpi-image-gen #55: Feature request: support for ZFS

VeloAddict, 2025-05-09:
It would be great to be able to install the system on ZFS.
You ask why? Before a system update is installed a new boot env is created, so if something fails you can always go back to the previous boot env.

:)

VeloAddict, 2025-05-10:
As I read in the docs, BTRFS is supported, why not ZFS?
Having a fat32 or ext4 /boot partition, and zfs pool on another partition, mounted to /, as rootfs.
And using ZFSBootMenu to boot from the ZFS partition and handle boot environments.
So flashing this image will reasult in a system that has a zfs rootfs with boot environments that allows rollback in case of something going foobar.

learmj, 2025-05-12:
> BTRFS is supported, why not ZFS?

btrfs is supported by genimage. zfs is not. rpi-image-gen uses genimage to create the filesystem images.

closed 2025-06-09 by learmj