raspberrypi/rpi-image-gen #56: Networking issue

originaltrini0, 2025-05-11:
Hello:

My target hardware:
CM5 4GB Lite
CM5 I/O board with an NVME disk

Thanks for this project!

I have been using Ubuntu to create headless servers using RPis. But I now would like a slimmed-down image to replace Ubuntu.
Initially, I tried using the `example/slim` example as a base for my build. Unfortunately, the wired network doesn't work.
I'm familiar with netplan, but I believe Debian uses `/etc/networking` by default. This directory does not exist.
I decided to simplify my project and build up from there.
Here is my external directory:
```
tree project/
project/
├── my.options
```

The `my.options` file only has `device_user1` and `device_user1pass` set.
I ran the build process with: `./build.sh -o ~/project/my.options`

This still results in an image where networking is not working.
Could you suggest why the wired network is not working?

Thanks

learmj, 2025-05-12:
Hi @originaltrini0 Thanks for your report. The minimal rpi-image-gen install uses systemd to configure the network. If the kernel/udev enumerates your wired device network interface as eth0, file ```meta/sys-apps/systemd-net-min.yaml``` should provide you with a valid systemd fragment installed at ```/etc/systemd/network/01-eth0.network```. If networking isn't operational for you, I suspect your netif isn't eth0.

originaltrini0, 2025-05-12:
Thanks for getting back to me @learmj.

I switched back to using my external directory based on `example/slim` and changed my config class to `cm5`.
I will have to test the pi5 class again, but now I am happy to report that networking is working.
But what is not working out of the box is DNS resolution. A quick edit of `/etc/resolv.conf` fixed that. I will have to figure out why it didn't pick up my DNS servers via DHCP, or include a script to configure the file how I want during the build process.

Thanks!

closed 2025-05-12 by originaltrini0

learmj, 2025-07-24:
The example template has this:
```
[DHCPv4]
RoutesToDNS=false
```
...which was added because of an upstream bug in systemd that was supposedly fixed but which I still observed.
If you remove this ```/etc/systemd/network/01-eth0.network``` are things any better?

schmurfy, 2025-07-29:
> The example template has this:
> 
> ```
> [DHCPv4]
> RoutesToDNS=false
> ```
> 
> ...which was added because of an upstream bug in systemd that was supposedly fixed but which I still observed. If you remove this `/etc/systemd/network/01-eth0.network` are things any better?

it works, thanks !