raspberrypi/rpi-image-gen #57: Documentation

Choccy-vr, 2025-05-11:
I am struggling to get a hang of this. I think some good documentation would be really helpful. I couldn't really find all of the options I could put in my config file in the readme. Also might just be me I find it easier to read and understand the info if it was on its own separate page and sectioned on a docs page.

learmj, 2025-05-15:
Hi. The readme specifically mentions several files which contain settings and config options. Those are the ones to look at. The readme also describes how files are read and settings aggregated together.

I think there's probably always 'improvements' that can be made to most documentation, but quite often those are subjective and a matter of opinion. I hoped that between the readme, the in-tree documentation (like the .defaults files) and the examples, there would be enough for users to use to get to grips with.

If you feel an element of the documentation is missing or lacking specific details, please let me know what. It's important that the tool is usable and I'll happily make modifications to the readme to improve usability.

learmj, 2025-05-15:
Also: as more the tool evolves, it's likely that the config options will be extended. Having them nested in files isn't great from a user perspective. I appreciate that. There may be things we can do to improve things there (eg git book).

Choccy-vr, 2025-05-15:
All I am saying is that a central wiki with all the existing information that is segmented among all the different folders would be really helpful

daybyter, 2025-05-22:
Same here. At the moment I am struggling to set a fixed IP address for my image. I am just putting config files in my rootfs overlay, which seems to be the wrong way to do? Maybe meta-hooks/systemd/netgen should be modified?

learmj, 2025-05-23:
What's the name of your network interface?

daybyter, 2025-05-26:
I don't know at the moment. I generated the webkiosk image and it starts at least the browser. But cannot connect to the outside and I cannot ssh into the pi at the moment, so cannot look at the logs at the moment.

learmj, 2025-06-13:
Do you have ```IGconf_device_user1pass``` set? Check ```work/<image name>/artefacts/config.yaml``` to see the entire recipe used for your build. If ```IGconf_device_user1pass``` is not set to a string, the user1 account will be locked for password based logins (on any tty or ssh).
However, you will be able to login via ssh without setting ```IGconf_device_user1pass``` **if** you enable ssh public key auth. This is an example which I regularly use:
```
IGconf_meta_ssh_pubkey_user1=$(< ~/.ssh/id_rsa.pub) IGconf_device_ssh_user1=y ./build.sh <args>
```
Login on any tty is not possible but login via ssh works fine.
Setting ```IGconf_device_ssh_user1=y``` is fine and will work, but only if you have a passwd set for user1.
See https://github.com/raspberrypi/rpi-image-gen/blob/master/meta/net-misc/openssh-server.defaults

However, if you're getting connection refused, this implies the sshd service is not running.

daybyter, 2025-06-13:
> For ssh you need to set `device_ssh_user1=y` in `examples/setoptions/my.options` or similar.

Changed the options to:
====
# Comments are supported
DEVICE_HOSTNAME=frobulator
# All variables will be translated to lower case and made available to hooks
device_user1=pilot
dEvIcE_UsER1pASs=removeme
ssh_device_user1=y
# Variables set in the .cfg file can be overridden
IMAGE_NAME=mydevice
====

No login possible.

learmj, 2025-06-16:
> So ssh user1 is enabled?

```
# If y, automatically include net-misc/openssh-server in the profile to enable
ssh_user1=n
```
The variable set by the above (IGconf_device_ssh_user1)  governs whether net-misc/openssh-server is _automatically_ pulled in and if the sshd service is started on the device. It's set to n in your output snippet.

Variable ```IGconf_ssh_device_user1``` also in your output snippet, but is not a variable set or used by rpi-image-gen.

> What irritates me a bit is, that the check for the user1pass is before you see the set password?
> Line 87 is the check, but line 144 shows the password.
> 

I think you're misunderstanding the output in ```config.yaml```. Not all lines in this file are executed in the order they are shown. The variables under the ```env``` section are present in the _environment_ when bdebstrap runs. Sections named ```<foo>-hooks```, e.g. ```customize-hooks```, do have their lines executed in order. When the YAML is merged together, all these operations are aggregated in the order thy YAML layers are read. So it doesn't matter that the password can be seen in a variable in this file after the script line showing it being used. These variables are all set before bdebstrap even runs.

> What also irritates me is, that a lot of the other stuff has stopped working. The image is no longer resized to sdcard size as an example.

Sorry to hear you're irritated, but nothing rpi-image-gen does performs any resizing. I guess this is in custom logic / additions.

daybyter, 2025-06-16:
The variable set by the above (IGconf_device_ssh_user1) governs whether net-misc/openssh-server is automatically pulled in and if the sshd service is started on the device. It's set to n in your output snippet.

So I added 

ssh_user1=y

to ./examples/setoptions/my.options

, build the webkiosk example with

./build.sh -c generic64-apt-simple -D ./examples/webkiosk/ -o ./examples/setoptions/my.options

and look at 

work/mydevice/artefacts/config.yaml

And find:

IGconf_device_ssh_user1: n

So what do I have to add to make user1 a ssh-user?

stu-spp, 2025-06-16:
The option is `device_ssh_user1` not `ssh_user1`.

You must set `device_ssh_user1=y` in `examples/setoptions/my.options` not `ssh_user1=y`.

learmj, 2025-06-17:
@daybyter Please refer to:
https://github.com/raspberrypi/rpi-image-gen?tab=readme-ov-file#input-parameter-assembly

When the tool loads settings from the options file, it adds prefix ```IGconf``` to all variables.
When the tool loads settings from a file such as device.defaults, it adds prefix ```IGconf`_device`` to all variables.

daybyter, 2025-06-17:
Yes! Login works again! Thanks a ton to @stu-spp and @learmj !

Still lots of problems, but at least I can login to the device now and check for errors!

Thanks again!

Choccy-vr, 2025-08-16:
Is there new documentation coming now because of the new update I saw something's that it might be coming

closed 2025-09-05 by learmj