raspberrypi/rpi-image-gen #58: How to consume all available disk space?

originaltrini0, 2025-05-12:
Hello:

How can I tell rpi-image-gen to use all the disk when partitioning the drive (nvme in my case)?
Here is what I defined in my config file:
```
[device]
class=mypi
profile=pi5

[image]
layout=mbr/simple_dual
boot_part_size=200%
root_part_size=300%
name=deb12-pi5
```

Here is what disk usage is like:
```
lsblk -l
NAME      MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
nvme0n1   259:0    0 119.2G  0 disk
nvme0n1p1 259:1    0    88M  0 part /boot/firmware
nvme0n1p2 259:2    0   1.3G  0 part /
```

```
df -h /
Filesystem      Size  Used Avail Use% Mounted on
/dev/nvme0n1p2  1.3G  402M  782M  34% /
```

It used 1.3 GB out of the 128 GB.
What does 100% of the root|boot_part_size represent?

Thanks

stu-spp, 2025-05-12:
One way to fix this is to (at some after first boot):
```
sudo parted /dev/nvme0n1 resizepart 2 100%
sudo resize2fs /dev/nvme0n1p2
```

originaltrini0, 2025-05-12:
Ok, I thought there was a built-in flag that did this.
I'll make sure to create a script to do what you recommended.

Thanks

closed 2025-05-12 by originaltrini0

learmj, 2025-05-13:
Hi. Thanks for your report.
[https://github.com/raspberrypi/rpi-image-gen/tree/master/image](https://github.com/raspberrypi/rpi-image-gen/tree/master/image) should help to explain the sizing units, or at least point to the source of where further information can be obtained. 100% is basically a filesystem that is 100% full. 200% will give you a filesystem that has 50% free (ie it's created twice the size of the data it contains). Percentage sizing is approximate and relative to the total size of data used to fill the (partition) filesystem image.

There isn't any support for on-device resize of filesystems in rpi-image-gen. Unlike Raspberry Pi OS (which does support this) rpi-image-gen currently has a more focused view of creating images specific to the device, including storage. As @stu-spp mentions, it's easy to add your own 'first time' boot service script to perform on-device resize but as far as our provisioning tools go and the roadmap there, we recommend using sparse images and the [pi-gen-micro](https://github.com/raspberrypi/pi-gen-micro) fastboot gadget to write the images directly. Other tools like bmaptool offer similar flexibility, ie being able to sparse down an image with lots of free space and write it to the device without the need to perform any on-device resize.