raspberrypi/rpi-image-gen #59: Missing example: build full Raspberry Pi OS Bookworm Desktop (not minimal) with rpi-image-gen

sincethestudy, 2025-05-14:
Hello. I currently use the default Raspberry Pi OS that is recommended in the RPI Imager (screenshot below). I'd like to use this tool to create a custom OS based on that, but it seems all of the examples and even the default build creates like a really minimal image that is missing a lot of stuff, and im having trouble just replicating the basic desktop features.

Could you please make an example that matches the build for the default image, this would be really great because I presume a lot of people are like me and use that one as their base image!

[image]

Thanks!

LorandMarton, 2025-05-14:
@sincethestudy This might help answer your question - I had a similar one myself.
 [https://github.com/raspberrypi/rpi-image-gen/issues/29#issuecomment-2717049724](https://github.com/raspberrypi/rpi-image-gen/issues/29#issuecomment-2717049724)

I checked which packages are added in [pi-gen Stage 2](https://github.com/RPi-Distro/pi-gen) and added those to my config. :)

learmj, 2025-05-14:
Hi @sincethestudy 
There is work ongoing to enable this functionality and we do intend on providing infrastructure in rpi-image-gen to allow users to create a system using a more modular approach using the layer scheme that exists today. The intention is to have groupings of layers which can be built up via a profile resulting in an image which has a given set of features. I envisage having profiles for a lite image, minimal desktop image, etc. It's early days but it is something we intend on rolling out.

> I'd like to use this tool to create a custom OS based on that, but it seems all of the examples and even the default build creates like a really minimal image that is missing a lot of stuff

That's right, but it's intentional because the goal of rpi-image-gen is to allow you to 'build your image your way' from the bottom up. I totally understand that people might like to have a lite image with a subset of features, or a full-fat desktop image. It would be nice if that was supportable in rpi-image-gen. It all comes down to the granularity of the layer grouping. You can view such a grouping as a "_developer-selected, curated software suite that generally relates to a certain family of related software packages_" - Gentoo linux adopts that description for it's _meta ebuilds_ and you can view the rpi-image-gen groupings in a similar way.

learmj, 2025-05-23:
One issue per topic is the normal procedure.

devopsbob, 2025-08-13:
Hi @learmj , thank you for posting the update. I hope to give this a retry in mid-October. Please post any repo/example link to help kickstart (me) again...

learmj, 2025-09-05:
It's highly unlikely that rpi-image-gen will gain support for generating a Bookworm desktop image, but it will gain support for generating a Trixie Desktop image. This ticket is specifically about the former, so I'm closing it. A new ticket requesting  Trixie Desktop support is welcome.

closed 2025-09-05 by learmj