raspberrypi/rpi-image-gen #61: Potential bug parsing custom profile files

originaltrini0, 2025-05-14:
If I defined my profile without a blank newline, `my/systemd-services` is never discovered:
```
debian/bookworm/arm64/base-apt
rpi/debian/bookworm/apt
rpi/misc-utils
rpi/boot-firmware
rpi/arm64/linux-image-2712
rpi/user-credentials
rpi/misc-skel
sys-apps/systemd-net-min
my/essential
my/systemd-services
```

If I define the file with a trailing blank newline, it works as advertised:
```
debian/bookworm/arm64/base-apt
rpi/debian/bookworm/apt
rpi/misc-utils
rpi/boot-firmware
rpi/arm64/linux-image-2712
rpi/user-credentials
rpi/misc-skel
sys-apps/systemd-net-min
my/essential
my/systemd-services

```

Feel free to let me know if you need anything else from me.

Thanks

learmj, 2025-05-14:
Hi @originaltrini0 
Thanks for the report.
The default config ```config/generic64-apt-simple.cfg``` pulls in profile ```bookworm/apt-min64```, and that profile has the following as it's last line:
```sys-apps/fake-hwclock```

That layer pulls in ```meta/sys-apps/fake-hwclock.yaml``` which installs ```fake-hwclock``` and runs some scripts, one of which enables the fake-hwclock systemd service.

When I run the default build, I see:
```
enable-units: Enabling unit fake-hwclock...
enable-units: Calling chroot "/path/to/work/deb12-arm64-min/rootfs" /bin/systemctl enable "fake-hwclock"
Synchronizing state of fake-hwclock.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable fake-hwclock
```
This can only happen if the layer was loaded, which implies the last line was read from the profile.
Therefore I conclude that there isn't anything wrong in the env I'm running the build in or the handling of the profile in that environment.

What editor / IDE are you using to edit your files? What are the line endings?
What do you see if you do:
```
cat -e profile/bookworm/apt-min64
```
?
Do all lines end with ```$```?

originaltrini0, 2025-05-14:
Thanks for getting back to me @learmj.

I am using a MacBook (ARM)/VSCode, and VSCode is currently set to LF line breaks.

Here is the output you asked for. This is with no empty newline:
```
imagegen@48e35c7cbebe:~/rpi-image-gen$ cat -e ~/project/profile/pi5
debian/bookworm/arm64/base-apt$
rpi/debian/bookworm/apt$
rpi/misc-utils$
rpi/boot-firmware$
rpi/arm64/linux-image-2712$
rpi/user-credentials$
rpi/misc-skel$
sys-apps/systemd-net-min$
my-apps/essential$
my-services/systemd-servicesimagegen@48e35c7cbebe:~/rpi-image-gen$
```

Thanks

learmj, 2025-05-14:
Yup. Your last line doesn't have an end of line character (indicated by a dollar). Thats why it doesn't get read by the parsing code.
I guess this is caused by a vscode setting. A quick search indicates that you may be able to change this via 'preferences open settings'. But it isn't an rpi-image-gen bug.

originaltrini0, 2025-05-14:
Ok, let me play around with VSCode settings.

Thanks

closed 2025-05-14 by originaltrini0