raspberrypi/rpi-image-gen #62: Change Kernel Version

Choccy-vr, 2025-05-17:
I need to change my kernel version from 6.12.25 to 6.6.51 because of a driver. How can I change the kernel version to that do I need to downgrade my host kernel or just add a config line?

learmj, 2025-05-19:
Hi. If the kernel package is in apt, you can simply add the package as part of a layer. Two simple options are:
1) Edit the layer currently pulled in by your profile and add the package (eg change ```meta/rpi/arm64/linux-image-2712.yaml``` to contain the package you want rather than ```linux-image-rpi-2712```.
2) Add a new layer, define the packages you want in it, and pull that in from your profile (eg if using ```profile/bookworm/apt-min64``` replace ```rpi/arm64/linux-image-v8``` with your new layer).

I'm currently working on device class specific layers (sneaky peak [here](https://github.com/learmj/rpi-image-gen/tree/device_class_profile)). This is lower priority than other work I'm doing at the moment, and it won't be merged until all device classes in the tree can be supported, but it will enable you to generate an image specific to the device you're targeting without affecting other devices classes.

LorandMarton, 2025-05-19:
@learmj For the btrfs filesystem I had to add an additional customize script, hook to run this command `chroot $1 update-initramfs -c -k 6.6.20+rpt-rpi-v8`
So customize10-initramfs will run and my customize11-init-update too. If I don't add my customize script then the chroot $1 update-initramfs -u willsilently fail.

learmj, 2025-05-19:
```scripts/bdebstrap/customize10-initramfs``` should always run at the end of all user packages / scripts and rebuild the initramfs for all installed kernels. That's what it's for (ie so the user doesn't have to do what you describe).
If it's not working like that for you, then please create a new issue and describe in detail what you're doing.

> For the btrfs filesystem

I don't see how what you've described can be related specifically to a btrfs filesystem.
We had https://github.com/raspberrypi/linux/issues/6436 but that was related to rpi-update.

closed 2025-08-05 by Choccy-vr