raspberrypi/rpi-image-gen #63: Add layer for running only from RAM

septatrix, 2025-05-19:
To improve longevity of the storage medium it is often desirable to run entirely from RAM (as long as the software is small enough and does not consume a lot of memory). In the wild there seem to be a few tools to convert a running image to one which run primarily from RAM but it would be favorable if one could built an image with such a setup from the start

learmj, 2025-05-23:
I agree it would be nice to have an example to create a small initramfs only based image.

In the interim, if you need this functionality, take a look at our https://github.com/raspberrypi/pi-gen-micro

tdewey-rpi, 2025-06-03:
> > In the interim, if you need this functionality, take a look at our https://github.com/raspberrypi/pi-gen-micro
> 
> Our image is a bit bigger as it also needs to include a GUI. Is there a limit on the initrd or boot partition size from the point of the RPi bootloader/firmware (for RPi 4)?

Yes - circa 96MiB for the initramfs.

You can indeed build bigger things with pi-gen-micro, but it's more involved (and with a bigger focus on creating minimal-footprint images). In the future, we may look to introduce examples in rpi-image-gen, but there is a definite intent to migrate our embedded Raspberry Pi Imager to pi-gen-micro, which may serve as a template on a shorter timeframe.

Have you contacted applications@ about your design? They may be able to offer greater advice & assistance.

tdewey-rpi, 2025-06-03:
> > > Our image is a bit bigger as it also needs to include a GUI. Is there a limit on the initrd or boot partition size from the point of the RPi bootloader/firmware (for RPi 4)?
> > 
> > 
> > Yes - circa 96MiB for the initramfs.
> 
> Okay, I fear that is not enough so will likely need a minimal initramfs which creates a ramfs and loads everything from the normal partition

Yes, I'd suggest use pi-gen-micro for that (take inspiration from the `cryptroot` image), and give it the requisite instructions for mounting & pivoting as you require.

> 
> > You can indeed build bigger things with pi-gen-micro, but it's more involved (and with a bigger focus on creating minimal-footprint images). In the future, we may look to introduce examples in rpi-image-gen, but there is a definite intent to migrate our embedded Raspberry Pi Imager to pi-gen-micro, which may serve as a template on a shorter timeframe.
> 
> Okay now I am a bit confused about the plethora of tools. rpi-image-gen is quite new and there are already plans to migrate away from it? There are pi-gen, rpi-image-gen, and pi-gen-micro...and they are all supported, actively developed and not deprecated?

Negative, the plans are to migrate _to_ rpi-image-gen, but we've had a range of other tools that were created for specific purposes. As time progresses, we hope to reconcile these other tools into rpi-image-gen. In practice, I believe today that you can effectively use rpi-image-gen in place of pi-gen, however pi-gen-micro fills a niche that is still unsatisfied by rpi-image-gen.

> 
> For context: I am currently looking into using [mkosi](https://github.com/systemd/mkosi) to build RPi images as it has a phenomenal UX and I have already been using it for other projects with great success.
> 
> > Have you contacted applications@ about your design? They may be able to offer greater advice & assistance.
> 
> No I haven't. Will look into that - I assume just a rough outline of the use-case is enough to get started?

Yes, that's where I'd begin. If they require further information, they'll ask.