raspberrypi/rpi-image-gen #70: Clean install - some warnings

jchidley, 2025-07-10:
Using a Pi5, nvme

A few warnings came up:

W: Download is performed unsandboxed as root as file /home/jack/rpi-image-gen/work/deb12-slim/rootfs/var/lib/apt/lists/partial couldn't be accessed by user _apt

jchidley, 2025-07-10:
It's not a massive problem - after all the people using this should be reasonably expert - but, as I said, it was unexpected. You know users are going to call you about it.

This is what Claude code (worth every penny that I spend on the MAX 20x subscription!) had to say about the errors:

```
## Common Build Warnings

The following warnings are normal and expected during the build process:

### APT Sandbox Warning

W: Download is performed unsandboxed as root as file /path/to/rootfs/var/lib/apt/lists/partial couldn't be accessed by user _apt

This occurs because the build runs in a containerized environment where APT's sandboxing features are automatically disabled for compatibility. This doesn't affect the build quality or security of the final image.

### /proc Mount Warning

W: since mounting /proc normally failed, /proc is now bind-mounted instead

This is a limitation of rootless container builds. The build system works around this automatically and it doesn't impact the final image.

### Compression Warning

W: No zstd in /usr/bin:/sbin:/bin, using gzip

This appears when zstd compression isn't available in the build environment. The tool falls back to gzip compression. Installing dependencies with `sudo ./install_deps.sh` ensures zstd is available.
```

closed 2025-07-10 by jchidley

learmj, 2025-07-10:
> This appears when zstd compression isn't available in the build environment. The tool falls back to gzip compression. Installing dependencies with `sudo ./install_deps.sh` ensures zstd is available.

Well, your AI is partially correct here, but not completely accurate...

You'll see that the deps are auto checked by the build process, so you'll have zstd installed on the host (ie on your Pi5) but it won't be installed in the chroot (where the initramfs is built).
Invocation of ```install_deps.sh``` affects what's installed on the host - not in the chroot.

Thanks a lot for the feedback and for the taking the time to try rpi-image-gen. The tool is very much in its infancy and is deficient in various areas, but we plan to address things as time goes on. It will be receiving some major updates soon.

stu-spp, 2025-07-10:
Can I get some more context about this comment mentioning that options file is going away?

> FWIW, support of the options file will be going away. All configuration will be held in the config file or provided via the command line (or combo of both). The options file is a hacky way to inject config variables into the flow, but it's confusing and detracts from how we'd like users to get onboard with the tool.

Can I get a migration guide or some documentation? When is the options file planned to be deprecated?

Where is this new config file?  What options will be injected via the command line?

learmj, 2025-07-10:
Hi @stu-spp 
Ideally, I would like to be able to inform users of upcoming compatibility breaking changes because I hate inconveniencing users of a tool that I'd like them to stick with, and improve for them.

The problems with supporting an options file are:

- It hasn't been clear to all users what it's for, or how to use it ( and it's a bit of a side-channel hack)
- Usage of the options file is not subject to error checking (ie you can set variables in it that can easily be ignored which costs time and causes frustration, eg #57).
- There is no structure to what's in it other than ```<known prefix>_<variable>```

There have been a few tickets where I've felt that by supporting variables provided by config, env and options it's simply too many sources of input. The way I see it going is:

- Config file provides the core configuration
- Command line can include any number of key=value pairs, e.g. ```IGconf_device_class=cm5```

Specifying variable assignments on the command line will take precedence over config. This makes usage of that intuitive (eg ```make FOO=bar```) but taking IGconf variables from the environment will also not be supported because it's dangerous and can easily cause unintended pollution.

Migration guide is simple:
If your options has ```foo_myvar=x``` you can either have...
```
[foo]
myvar=x
```
...or pass ```IGconf_foo_myvar=x``` on the command line like ```./build.sh <args> -- IGconf_foo_myvar=x```.

That's how I intend on it shaking out.

Config files have always existed (eg ```config/```), which kind of proves my point that something somewhere needs to change with how folks are using (or expecting to use) the tool :-)