raspberrypi/rpi-image-gen #75: Image geometry wrong?

cmahnke, 2025-07-28:
I tried to use `parted` on a generated image ant it fails:
```
parted -s /app/images/deb12-arm64-min.img unit s print
Error: Can't have a partition outside the disk!
Model:  (file)
Disk /app/images/deb12-arm64-min.img: 2735424s
Sector size (logical/physical): 512B/512B
Partition Table: unknown
Disk Flags: 
```

Running `fdisk` on the fie gives:
```
root@dadbc2cdeab7:/app# fdisk -l /app/images/deb12-arm64-min.img
Disk /app/images/deb12-arm64-min.img: 1.3 GiB, 1400537088 bytes, 2735424 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                           Boot  Start     End Sectors  Size Id Type
/app/images/deb12-arm64-min.img1 *     16384  196607  180224   88M  c W95 FAT32 (LBA)
/app/images/deb12-arm64-min.img2      196608 2736127 2539520  1.2G 83 Linux
```

The file seems to have less sectors in total (2735424) then the end of the second partition suggests (2736127).

Does anyone knows what's going on?

cmahnke, 2025-07-28:
This is a work around:
```
ACTUAL_SECTORS=$(fdisk -l "$RPI_IMAGE_PATH" | head -1 | awk '{print $7}')
EXPECTED_SECTORS=$(fdisk -l "$RPI_IMAGE_PATH" | tail -1 | awk '{print $3}')
MISSING_SECTORS=$((EXPECTED_SECTORS - ACTUAL_SECTORS + 1))

if [ $MISSING_SECTORS -gt 0 ]; then
    echo "Padding by $MISSING_SECTORS"
    dd if=/dev/zero bs=$(echo "512 * $MISSING_SECTORS" | bc) count=1 >> "$RPI_IMAGE_PATH"
fi
```

learmj, 2025-07-28:
Hi. Thanks for your report. Newer genimage supports ```fill``` where it will fill up to the end of the requested ```size```, but the current genimage in Debian bookworm (my base testing platform) is not new enough to support this, so the option is not used in the in-tree genimage templates. There is padding logic currently in place (see #51) so the image should always be a multiple of the sector size, but this doesn't explain your problem.
However, neither of these would explain why the partition table in the image indicates that (presumably) the last partition is smaller than it is.
What system are you building on, and what version of genimage is your build using?
To debug issues like these further, please attach your final genimage.cfg from the output dir.

cmahnke, 2025-07-28:
I'm building "nothing" (the default) running just `./build.sh` on a Docker image based on `debian:bookworm-slim`

```
image deb12-arm64-min.img.sparse {
   android-sparse {
      image = deb12-arm64-min.img
   }
}

image boot.vfat.sparse {
   android-sparse {
      image = boot.vfat
   }
}

image root.ext4.sparse {
   android-sparse {
      image = root.ext4
   }
}

image deb12-arm64-min.img {
   hdimage {
      align = 8M
      partition-table-type = "mbr"
   }

   partition boot {
      in-partition-table = true
      partition-type = 0xC
      image = boot.vfat
      bootable = true
   }

   partition root {
      in-partition-table = true
      partition-type = 0x83
      image = root.ext4
   }
}

image boot.vfat {
   vfat {
      label = "BOOT"
      extraargs = "-S 512 -i 1171769D"
   }
   size = 200%
   mountpoint = "/boot/firmware"
   exec-pre = "'/macguffin/rpi-image-gen/image/mbr/simple_dual/setup.sh' BOOT 1171-769D edb3a12d-8dcb-496d-b644-c17f44396421"
}

image root.ext4 {
   ext4 {
      label = "ROOT"
      use-mke2fs = true
      mke2fs-conf = '/macguffin/rpi-image-gen/image/mbr/simple_dual/mke2fs.conf'
      extraargs = "-U edb3a12d-8dcb-496d-b644-c17f44396421"
   }
   size = 300%
   mountpoint = "/"
   exec-pre = "'/macguffin/rpi-image-gen/image/mbr/simple_dual/setup.sh' ROOT 1171-769D edb3a12d-8dcb-496d-b644-c17f44396421"
}

```