raspberrypi/rpi-image-gen #78: Trixie

Choccy-vr, 2025-07-30:
I was trying to build an image of Trixie (no customization other than username and pass) and it keeps erroring saying that adduser is not found. I know that bookworm works so I don't think it is a problem with my system.
```
sh: 1: adduser: not found
E: setup failed: E: command failed: chroot $1 sh -c "if ! id -u $IGconf_device_user1 >/dev/null 2>&1; then adduser --disabled-password --gecos \"\"  $IGconf_device_user1; fi"
I: main() received signal PIPE: waiting for setup...
E: mmdebstrap failed to run
2025-07-30 17:44:37,003 bdebstrap ERROR: mmdebstrap failed with exit code 29. See above for details.
```

learmj, 2025-07-31:
I think it's probably likely that ```adduser``` has been removed from the ```variant: apt``` of the trixie base, which the default profile uses. I saw this not so long ago for other packages (login). Adding the package to the base layer fixes it for me, ie
```
diff --git a/meta/debian/trixie/arm64/base-apt.yaml b/meta/debian/trixie/arm64/base-apt.yaml
index f867926..7bd06f0 100644
--- a/meta/debian/trixie/arm64/base-apt.yaml
+++ b/meta/debian/trixie/arm64/base-apt.yaml
@@ -13,6 +13,7 @@ mmdebstrap:
   packages:
     - login
     - linux-sysctl-defaults
+    - adduser
   essential-hooks:
     - echo tzdata tzdata/Areas select "$IGconf_device_timezone_area"
         | chroot $1 debconf-set-selections
```

learmj, 2025-07-31:
FWIW I think the correct fix is to have adduser installed by the layer that performs the user admin, ie ```user-credentials.yaml ```.
fyi - There are some big changes coming to rpi-image-gen soon.

benlongo, 2025-08-15:
I'm still seeing this on `main` - @learmj re. your comment about big changes, would you want a PR fixing this or should it wait?

learmj, 2025-08-16:
It's fixed in the changes that are coming. I'm hoping to merge those changes by the end of the month. Temporary solution is to just add adduser to the pkg list in the rpi-user-creds layer.

learmj, 2025-08-19:
Thanks @stu-spp I hear you.

learmj, 2025-08-20:
> Thanks @stu-spp I hear you. 

Hi @stu-spp Please can you create a new issue for this. There are probably implementation details that need discussing. Thx