raspberrypi/rpi-image-gen #79: Updates and breakages Q3'25

learmj, 2025-08-13:
rpi-image-gen continues to evolve and while the tool is still very much in its infancy, it has become clear that the philosophy behind its use needs focus and refinement. Several users have expressed confusion and frustration about how to customise their build using the config variables, and functionality one would reasonably expect from a 'build system' is missing (e.g. clean). To try and address these concerns I'm making the following major changes to rpi-image-gen. Some of them do break compatibility with how the tool is currently used, and I apologise for that. I'd really rather avoid causing inconvenience to users of a tool I'd like them to stick with. I believe these changes are for the greater good and will improve the experience for all, as well as making the tool much more powerful, easier to use and smoother for new users to get onboard with.

**What's changing?**

1. Support of the options file has been dropped. The options file was badly documented, subject to zero checking and was effectively a side-channel hack to inject variables into the build. It hasn't been clear to all users what it's for, or how to use it. Lack of variable checking costs time and causes frustration (eg #57). There is no structure to what's in the options file other than ```<known prefix>_<variable>```.

Variables will only be able to be supplied:

a) In the config file
b) On the command line
A combination of both is supported.

Option file migration guide:
If your options file has ```foo_myvar=x``` you can either have...
```
[foo]
myvar=x
```
...or pass ```IGconf_foo_myvar=x``` on the command line.

2. Relocatable execution (advanced preparation for debian packaging). In the future, rpi-image-gen is envisaged to be packaged as a deb. This means it needs to self locate better and be able to resolve paths to internal assets.  

3. rpi-image-gen will be the name of the top level script and not build.sh.
 
4. Custom metadata support and consolidation of 'info files' into layers. This is the big one. rpi-image-gen now leverages a layer approach for everything - device, image, etc. Layers still adhere to the same bdebstrap YAML files but can now describe and declare variables using custom ```X-Env-``` metadata fields based on [DEB822](https://repolib.readthedocs.io/en/latest/deb822-format.html) syntax. Adoption of this makes the tool much more powerful and improves things across the board, allowing support for automatic variable validation, default values, dependencies between layers, inheritance, auto-generation of documentation, etc. New python classes constitute the 'core engine' and manage config file parsing, metadata and layer orchestration.

5. No more profiles. A profile is now a layer in its own right and uses metadata to define its dependencies. rpi-image-gen searches for all layers using a combination of built-in paths and source directory paths as it tries to locate layers defined by the config file. Config files now use the following, allowing the engine to manage all dependencies:
```
[layer]
foo=bar
```

6. The core engine establishes the build order of all layers, automatically pulling in dependencies as required. Dependency handling between layers completely removes the problem of, for example, layer A needing to install to ```/home/$USER``` (eg SSH keys) but the user, group etc not yet being created because that's done by layer B. Now, layer A can mark B as a dependency so it's ahead of it in the build order.

7. ```[profile]``` is no longer a section recognised by rpi-image-gen any more. Custom sections can still be added in the config file with their corresponding variables being parsed and included throughout the build env as ```IGconf_<section>_<key>=<value>```. The examples will all be updated to show this. 
 
8. Various configuration variables have been renamed, eg ```IGconf_sys_outputdir``` -> ```IGconf_image_outputdir```. Custom consumers of these variables will need to be updated.

9. Config file parsing now supports INI or YAML and has much improved include support. Similar to the layers, config files utilise a PATH based methodology for resolution. YAML is envisaged to be the default config file format in the future, simply to unify the format of all files. The core engine already has a .cfg -> .yaml migrate command built in to help users update. Files that used to define build options and defaults have been removed with all configuration information and variables residing in the YAML layers.

10. Automatic variable checking and validation is performed prior to starting build. The new metadata fields allow the engine to validate all variables understood by the layer chain as well as providing informative output about which variable has been set by which layer (#65 ) as well as an interactive flow allowing for pseudo dry run.

11. There is a set 'Policy' now where variables set by one layer can be overridden by another, forcibly set, skipped etc, all based around defined execution of the layer dependency chain and 'first/last wins' flow. This is all covered by new comprehensive documentation.

12. The layer metadata and supported fields lend themselves extremely well to automatic generation of documentation. This is a big win and will assist greatly with concerns similar to those raised in #57 . The layers are the single 'source of truth' for variables they define and require, and a layer's documentation can clearly describe these attributes as well as Debian specific attributes like repo mirror(s), suite, arch, packages, etc.

13. New CLI commands have been added to perform raw metadata introspection and validation as well as layer dependency information, listing, categories, validation, etc. Since layers are the key component of rpi-image-gen, it's imperative that developers have quality tooling support for working with them.

14. The tool will operate on a source directory (defaults to .). This essentially changes the philosophy of it to something more akin to a C compiler - it automatically knows where to go looking for built in libraries and headers (layers in this case). Your code (i.e your 'src directory') just contains the things that you want, and it references the built-in assets (layers). This makes it much easier for users to apply their own changes on top of the built-in / 'upstream' assets therefore making it much easier to simply add in the pieces you want on top of the 'substrate' provided by rpi-image-gen.

The above is a summary of the last few months of rpi-image-gen development effort and I'm hopeful the changes will help to make the tool a solid choice for developers. As before, feedback and suggestions are always welcome.

stu-spp, 2025-08-13:
Awesome! Thank you @learmj !!!

Where is your personal branch? I'd like to start using it.

So I currently build like this:
```shell
./build.sh -c config/trixie64-apt-simple.cfg -D custom-device/ -o custom-device/my.options
```

where custom-device has a simple bdebstrap `customize01` and `my.options` is a simple:
```ini
# Comments are supported
device_hostname=custom-device
device_user1=pi
device_user1pass=password
device_ssh_user1=y
image_name=bigmac2
```

How would I convert this to build in your branch?

```bash
./rpi-image-gen -c config/trixie64-apt-simple.cfg ...
```

learmj, 2025-08-13:
Hi @stu-spp Brave. I like it :) Branch is [here](https://github.com/learmj/rpi-image-gen/tree/flow_dev)
By the sounds of it, create custom-device/your.cfg and in it put:
```
!include trixie-minbase.cfg

[device]
hostname=custom-device
device_user1pass=<stronger passwd needed to pass validation!>

[image]
name=custom-device
```
The examples I'm updating adopt putting the config file file in ```<src dir>/config/``` but it's not mandatory.

The automatic pulling in of SSH support via a variable is gone. ```trixie-minbase.cfg``` pulls in layer [trixie-minbase](https://github.com/learmj/rpi-image-gen/blob/flow_dev/meta/suite/debian/trixie-minbase.yaml) which has openssh-server as a dependency, so you get SSH functionality out of the box.

On the device side, user pi is the default anyway so I removed that from the new config.

The variable validation stage will prevent an image being built with your password as it stands. The password needs to conform to [this](https://github.com/learmj/rpi-image-gen/blob/flow_dev/meta/device-base.yaml#L38).

Build with:
```
./rpi-image-gen build -c /path/to/custom-device/your.cfg -S /path/to/custom-device/
```
```-S``` sets the source directory to there so rpi-image-gen will look for layers in ```meta/foo.yaml``` if you chose to adopt the layer approach in stead of the bdebstrap script. I've already update the webkiosk example to do this.

All being well that will work and still trigger your script in ```/path/to/custom-device/bdebstrap/customize01```.

Let me know how it goes!

cmahnke, 2025-08-13:
That's a great proposal / outlook. My only concern is currently the drop of the options file. Have you considered switching to a `.env` approach like one would do with Docker or web projects (JS, PHP) ?
The benefit would be the separation between common configuration and secrets (like WiFi config)..

learmj, 2025-08-13:
> That's a great proposal / outlook. My only concern is currently the drop of the options file. Have you considered switching to a `.env` approach like one would do with Docker or web projects (JS, [...]

Thanks. There's nothing in the options file that you can't do in the config file. Support of the config file remains - only the options file has been dropped.

learmj, 2025-08-15:
> This sounds great!
> Personally I prefer in-repo documentation, f.e. md or adoc, since these are offline available, versionized, searchable with grep and they can be hosted as a website too.  
> But buggy table rendering sounds not great, any chance to improve this?  

Thanks. The html content is static so would be available offline, too. I can likely script up multiple formats.

learmj, 2025-08-20:
> It would be helpful for me if all possible entries in a config file would be listed and those currently not used are commented out. 

Not quite sure what you mean. Once the changes from #79 get merged, the variables which can be set depend entirely on what layers are pulled in. This is because it's only layers that declare variables (custom variables can still be specified in the config file and propagated through the build as they are today). If you mean will it be possible to indicate variables which take effect based on the layers that get pulled in, yes this is already done as part of the #79 changes on my branch. However, if there are variables in the config file that will get ignored by the layers that get pulled in, these are not highlighted. My logic for this is that if run interactivity, you can inspect the output of the tool to make sure the variable you've set in your config file have been applied. I've got new documentation written which mentions this. How does that sound?

> I would also prefer some releases of the tool, so one can write a project for it and it will run at any time, even if newer releases are available. Even if it is a 0.0.1Beta release. 

I'm happy to tag before the big merge if it helps anyone. I'll definitely tag after merging as it is a substantial uplift that certainly warrants a release version (there are no released versions right now). Debian packaging will come later but I plan to tag releases going forwards anyway, particularly when there are worthy/notable changes made to the tool. 

Sound ok?

daybyter, 2025-08-20:
A release would definitely help me a _lot_ . I am happy to stick to any version that works somewhat. I am not sure if you understand my point here. A release is important for me to protect my current work. Any major merge is a risk for me, since it will break my config. I have an expo coming up, where I would like to show a device with a generated image. So if the base of my work is constantly moving, it is almost impossible to build on it.

daybyter, 2025-08-20:
A release should have defined dependencies so it has a reproducable behavior?

closed 2025-09-05 by learmj