raspberrypi/rpi-image-gen #80: Proposal for examples

cmahnke, 2025-08-13:
In the last few weeks I was using this code to create some media stations for a exhibition. Some things were harder than the probably they should have been, but I was convinced by the approach and the speed of this coder (as compared to `pi-gen`).
From my experience I would like to see a few more examples addressing the following use cases:
* Integration with [Raspberry Pi Imager](https://github.com/raspberrypi/rpi-imager) : This tool allows setting WiFi credentials while writing an image, it would be great to habe an example doing this. Maybe even with WPA3 Enterprise
* Adding custom package sources (with https, trusted=yes etc).
* The old settings / variables have been misleading: For example setting a variable did't guarantee that the implied changes would be implemented since the user has been responsible for providing the required dependencies as well. One example (with the current code) is setting the console language, this won't change the keyboard layout by itself when `console-setup` hasn't been enabled via `systemd`...

Please keep in mind that there is #79 now, so this should target the result of it.

learmj, 2025-08-14:
Not sure I follow - variables are already namespaced to layers. If they weren't already they certainly are now. It's only layers that can declare variables once #79 lands. Can you perhaps provide an example for me to try and figure out how to address the problem you're experiencing?

learmj, 2025-08-14:
> That's the reason why I called it a possible feature request, im other words and the example above that would be:
> Defining `KEYBOARD_KEYMAP` should either make sure `console-setup` is setup (without having to the user handle dependencies) or throw an error. 
> I was just thinking about ways to implement this. If this is already solved with the new approach even better.

Ok. I think I see. You make a good point. However, it would seem to have a hard dependency on what should be (in terms of a modular build system) an optional way to init the device, ie what services to build into the image and start, have full-fat systemd init vs a thin busybox based init, while having the desired keymap.

Variables like device keymap aren't expected to enforce hard policy decisions about what packages to install - that's why their values are just pre-seeded in the debconf database. However, on the flip side, if you define a keymap you may expect the build system to do the needful, but that may involve it making hard decisions and those likely wouldn't suit all users.

It sounds like you're suggesting that systemd 
systemd-vconsole-setup.service should always be installed and started to configure the virtual terminals in order to honour the device keymap variable. I don't agree with that because a user may not care about the keymap variable at all and just keep it at the default, but definitely does wants a busybox system without systemd. To that end, by forcibly installing systemd just to honour this variable I'd be hardwiring systemd into everything. I'm not going to do that because a modular build system is about choice.

Maybe the best thing here (for this particular case) is to ensure that the systemd console service is started by any layer that installs systemd as the boot 'scheme' (eg layer systemd-net-min).

More generally, if there is a layer responsible for installing a 'sysv boot' it should try its best to honour variables like device keymap. 

Have you tried just starting the vconsole service in an image with systemd boot to see if it does what you need?

cmahnke, 2025-08-15:
Ok, I guess I see your point: Those variables have more the one implication I suspect? In this case it is not only are responsible for the setup of the console keyboard. The most obvious solution might be the namespaced duplication / specification of certain variables. Sticking with the example there might be something like `CONSOLE_KEYMAP` which would add as described above - that is `systemd` setup, including installation of required dependencies, but also sets the needed env vars. But due to the duplication (that is multiple specialised variables can set on global / general variable) that would require a validation since its' theoretically possible that those specialised variables result in conflicting values gor the global variable.

In terms of actual solution with possible (user) problems with the current code base, I guess we still aren't completely on the same page here: These are just suggestions. They are base on my experiences, but I don't expect solutions as part of this issue, actually I already solved this (for me) - I'm just a bit unhappy how it's currently solved...
For me it's just an example since I expect that this "trap" can also happen with others of the (currently) global variables - that's the reason why I argue to try to avoid them where they

cmahnke, 2025-08-19:
Just another idea for an example:
* The `kiosk` mode including the ability to use a on-screen keyboard of a touch screen

daybyter, 2025-08-20:
Thanks a lot for your proposals!

I am trying to build a similar application as you, it seems? 

You know the videos of MacMind? Like: 

https://www.youtube.com/watch?v=RnehTqAVqQQ

They helped me a lot, although it seems they are somewhat broken for me at moment with the pulled rpi-image-gen version.

cmahnke, 2025-08-20:
@daybyter That's definitelly possible...
I know @JonnyMacs videos.

These proposals for examples are hopefully the starting point for more: As I see it, the existing examples, new ones and configurations like the MacMind repositories on `rpi-image-gen` can be a starting point of a library of building blocks for a custom Raspberry Pie Image.

This issue will hopefully gather some more attention for ideas and maybe even implementations.

daybyter, 2025-08-20:
Before I forget: seems like the hint with executable customize* files was the solution in my case. Now the web-kiosk example seems to build including the kiosk service etc.

I wonder if it would be possible for rpi-image-gen to issue a warning if such a file was ignored? At best write it to a logfile... ;-)

learmj, 2025-08-22:
> @daybyter I think the bin/runner could theoretically warn when files with a matching prefix lack executable permissions.
> 
> https://github.com/raspberrypi/rpi-image-gen/blob/master/bin/runner
> 
> ```
> if [ -x "$script" ] ; then
> ```
> 
> matching scripts that fail this condition could be echo'd out. You might make a PR and see what @learmj thinks.

It's an easy and helpful thing to add. I'll roll it into the current changes.