raspberrypi/rpi-image-gen #83: ERROR: Kernel version unsupported -- Custom Kernel LOCALVERSION with '+'

JonahTsai, 2025-08-25:
The chance of using custom compiled kernel when using rpi-image-gen is high. At least for me, the whole point is generate a .img using a custom kernel for embedded device with everything the device needs all installed and configured, for non-technical folx. Can't ask those non-technical folx to install custom kernel, and other stuff. Of course, the other solution is to configure the whole thing then dump the whole SD card and distribute that. However, rpi-image-gen is a more repeatable way than that very manual process. I eventually will have custom kernel with statically linked module in it.

If you compile the Linux kernel, at the moment, it will append a '+'. to the kernel version. Supposedly, if you set the LOCALVERSION, it shouldn't append a '+', but at the moment it just will append a '+'. The only way I find could compile without a '+" on a branch without a release tag, say branch 6.16.y is to to comment out the echo '+' line in kernel scripts/setlocalversion.

The trouble with the '+' char appended to the kernel version is that when you run a "make -j15 .... deb-pkg", the generated .deb package name will have the '+' sign, and when you put that deb package to a custom layer in rpi-image-gen... the control script in the deb package will invoke /etc/kernel/postinst.d/*, which contains a z50raspberrypi hook script to update initramfs. Inside this hook script, there is a case v8, v8-rt, bcm2713, and v7, but with that '+' char, nothing matches, thus it falls through to the default case and gets you "ERROR: kernel version unsupported...."

AFAIK, the /etc/kernel/postinst.d/z50raspberrypi script comes from raspi-firmware package.

My current workaround is to comment out the Linux kernel script/setlocalversion's offending line. Perhaps Raspberry Pi could consider a better solution? Maybe add a case of v8+, v8-rt+, bcm2713+, v7+ in raspi-firmware package? Perhaps some even better mechanism to explicitly support custom compiled kernel?

XECDesign, 2025-08-26:
No plans to change the raspi-firmware package to handle custom kernels. It's meant to be used with officially provided kernels. For custom kernels, custom hooks should be used.

JonahTsai, 2025-09-13:
My main purpose for reporting the "issue" was so that others don't spend two weeks like me going down the wrong rabbit holes. Oh... and the rabbit holes I didn't know existed before! 

Until I found the root cause of that vague and very severe "ERROR: Kernel version unsupported" message, I couldn't be sure how "fatal" the error was, nor did I know what generated that error. Now that I found the root cause of it... And if by raising the issue, it could affect some changes, all the better. If not... saving others time would be enough. 

Still, would have saved me the two weeks, if the error message was clear on its origin.

I have since put a sed line in my Jenkins cross compiling build job to take out the '+' sign appended by the kernel build script, as a snot glue fixing another snot glue, since there doesn't seem to be any hope of getting raspi-firmware to fix it, and getting the upstream kernel to fix this is ridiculous. (Or I can write a custom script to fix raspi-firmware's postinst script, but the timing of the custom script could be tricky).  Sure, it's still snot, but it's my snot, and I know how to wipe my own snot. And I have no intention of publishing my Jenkins job for building my custom kernels. So, it won't send others down other rabbit holes. I will publish the resulting image, but not the sausage making details. I have gotten the custom kernel debian packages, including the dtb package and header package, handed over to another Jenkins job to build an .img for RPiZero 2Ws, running in USB gadget mode, with WiFI SSID, psk, hostname, SSHD turned on... all configured, with custom layers to install boot and cmdline.txt. I still yet to set up my own Nexus server to host the .deb artifacts. But, that can wait.

Thanks for looking into this.

XECDesign, 2025-09-14:
Yup, that's all valid and I'll make the message a bit more informative.

XECDesign, 2025-09-15:
Not sure how to quite capture the actual situation in a short one or two line message.

Incorporating some of @learmj's suggestions, I've downgraded the message to a warning and added a note:

```
    echo "WARNING: Unsupported kernel version ($kernel_version) - skipping setup"
    echo "NOTE: Manual boot configuration may be required"
```

However, that tells you nothing about where the message is coming from or what this manual boot configuration might be.

To actually grasp what's happening, you'd need to read https://kernel-team.pages.debian.net/kernel-handbook/ch-update-hooks.html at the very least.