raspberrypi/rpi-image-gen #84: Debugging `bdebstrap` / `mmdebstrap`?

cmahnke, 2025-08-25:
After trying to integrate self build Debian packages(`deb` files) I came to the conclusion that `bdebstrap` / `mmdebstrap` (hard to tell which one it is) is quite unfinished / buggy. See:
* [bdrung/bdebstrap#17](https://github.com/bdrung/bdebstrap/issues/17)
* [bdrung/bdebstrap#18](https://github.com/bdrung/bdebstrap/issues/18)

I already mentioned problems with HTTPS sources and `[trusted=yes]` here: https://github.com/raspberrypi/rpi-image-gen/issues/80#issue-3319368908

I' the end the tool is not only unfinished, but also hard to debug, since malformed is a broad term here, there is currently no way to find out what's the real issue:
* A (HTTPS) certificate can't be validated (which doesn't seem to work anyways)
* URL can't be used at all (404)
* Source not trusted 

```
E: Malformed entry 11 in list file /opt/rpi-image-gen/work/magisch-app/rootfs/etc/apt/sources.list (Component)
E: The list of sources could not be read.
E: apt-get update --error-on=any -oAPT::Status-Fd=<$fd> -oDpkg::Use-Pty=false failed
I: main() received signal PIPE: waiting for setup...
E: mmdebstrap failed to run
```

Since this isn't helpful at all, is it possible to get better messages?

closed 2025-08-26 by learmj

cmahnke, 2025-08-26:
Thanks, that is very frustrating and certainly another hint that that part of the toolchain doesn't feel very polished...

I've already created issues at bdebstrap, that's not what this issue was about, I was hoping to get better messages fro them into the output of `rpi-image-gen`. But given what you wrote I assume it won't be possible to pass `-o` to the underlying `apt` process... 

I don't need that kind of package authentication, it's just overhead, I'm building everything in one internal pipeline. It's just cargo cult security...

daybyter, 2025-09-09:
Is it correct, that the bdebstrap package is no longer available in debian trixie and the dependency check fails therefore?