raspberrypi/rpi-image-gen #90: IGconf_device_sector_size undefined when changing device layer from rpi5 to rpi4

paulsykes, 2025-09-09:
If I take one of the examples and update device layer from rpi5 to rpi4 in the config yaml, 'device-base' becomes no longer a dependency. As device-base is not included IGconf_device_sector_size is no longer defined and the image generation fails.
```
device:
  layer: rpi4
```

learmj, 2025-09-09:
Hi @paulsykes Yep. That's a bug and an artefact of simply dropping in the 'non family 5' device directories.... I'll fix.

learmj, 2025-09-09:
@paulsykes The linked PR fixes this problem. Apologies for the inconvenience. If you verify I'll regen the layer docs and merge.

closed 2025-09-10 by learmj

paulsykes, 2025-09-10:
You just beat me to it. Just tested with the kiosk example on rpi4 and is working. Thank you for your prompt assistance on this.

commit c059bca6 (2025-09-10): layer: rpi-generic64 should declare device-base as a dep

rpi-generic64 is a general purpose layer that supports all 64bit capable
devices. It should pull in device-base as a dependency. Doing so means
it can be used in isolation by device layers.

Fixes #90